gnss-sdr/gnss-sdr #763: > There are no specific settings. You can see PRN 32 tracked at B1I here: https://www.youtube.com/watch?v=BIf4bZthvUE And PRN 33, PRN 40, PRN 56 at B3I here: https://www.youtube.com/watch?v=jwFO56G27vk

pwang0112, 2024-03-24:
> There are no specific settings. You can see PRN 32 tracked at B1I here: https://www.youtube.com/watch?v=BIf4bZthvUE And PRN 33, PRN 40, PRN 56 at B3I here: https://www.youtube.com/watch?v=jwFO56G27vk

Anyways, I use the current code to process B1I observations, including code and carrier phase, and use rtklib to obtain a BDS B1I only short baseline RTK solution. I was not able to fix the interger ambiguity, although the accuracy of the baseline solution is at cm-level. did you try to use the B1I carrier phase for RTK?? 
Thanks

_Originally posted by @peiyzhou in https://github.com/gnss-sdr/gnss-sdr/issues/621#issuecomment-1166192550_

peiyzhou, 2024-07-15:
@pwang0112 Any updates? I also have this problem and still can't find the reason why it cannot conduct RTK ambiguity resolution   [https://github.com/gnss-sdr/gnss-sdr/issues/621#issuecomment-1166192550](url)

carlesfernandez, 2024-10-11:
Could you post your configuration file, so we will try to reproduce the problem?