gnss-sdr/gnss-sdr #766: GNSS-SDR on RHEL

db3134, 2024-04-02:
We would like to test GNSS-SDR for use in our systems. Unfortunately, we are limited to RHEL due to safety regulations, which means no Debian and no Ubuntu. Building from source does not comply with the safety guidelines either.

Are there any plans to publish GNSS-SDR in the rpm/dnf package manager?

db3134, 2024-04-02:
> What version of RHEL?

v8 or v9

> Is running a containerized version an option?

That would make things easier. Unfortunately not, I have checked that.

> If your constraints are really that strict, this may not be a mature enough tool for you.

You might be right with that one.

carlesfernandez, 2024-08-02:
There are no plans to release a rpm/dnf package in a short term. Could you elaborate more on the safety guidelines that the code does not meet?

closed 2024-09-13 by carlesfernandez