gnss-sdr/gnss-sdr #773: Linker error in Debug build: absl_log_internal_check_op DSO missing from command line

castle055, 2024-05-23:
I just forked the repository and was trying to setup a suitable development environment when I encountered a linker issue regarding abseil's logging library. Interestingly, this only happens when compiling the debug build.

The error in question is the following:

```
[100%] Linking CXX executable gnss-sdr
/usr/bin/ld: ../core/monitor/libcore_monitor.a(gnss_synchro.pb.cc.o): undefined reference to symbol '_ZN4absl12lts_2024011612log_internal17MakeCheckOpStringIPKvS4_EEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEET_T0_PKc'
/usr/bin/ld: /usr/lib/libabsl_log_internal_check_op.so.2401.0.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[3]: *** [src/main/CMakeFiles/gnss-sdr.dir/build.make:240: src/main/gnss-sdr] Error 1
make[2]: *** [CMakeFiles/Makefile2:2403: src/main/CMakeFiles/gnss-sdr.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2410: src/main/CMakeFiles/gnss-sdr.dir/rule] Error 2
make: *** [Makefile:712: gnss-sdr] Error 2
```

This seems to indicate that the `core_monitor` CMake target is not linking against `libabsl_log_internal_check_op`.

I don't have much experience with the abseil library, so I'm not sure if this is an environment issue on my part or if this is a bug in gnss-sdr's build system. However, adding `absl::log_internal_check_op` within the PRIVATE scope to the `target_link_libraries(core_monitor ...)` call in `src/core/monitor/CMakeLists.txt` seems to solve the issue:

```cmake
target_link_libraries(core_monitor
    PUBLIC
        Boost::headers
        Boost::system
        Gnuradio::runtime
        protobuf::libprotobuf
        core_system_parameters
    PRIVATE
        Boost::serialization
        absl::log_internal_check_op # <- Added this line
)
```

If this is a bug, I can submit a PR for it, but I'll need to check it doesn't conflict with any build options first because I suspect this is just a duck tape fix.

I'm compiling the `next` branch, specifically commit `f80c537` with system provided libabsl:
https://github.com/gnss-sdr/gnss-sdr/tree/f80c5373b78b5124e70d1e6743654a79145f0721

My system is running Arch Linux and abseil-cpp's binaries are installed under `/usr/lib/`:

```
$ ls -lsa /usr/lib/libabsl_log_internal_check_op*
 0 lrwxrwxrwx 1 root root    41 Apr 29 09:19 /usr/lib/libabsl_log_internal_check_op.so -> libabsl_log_internal_check_op.so.2401.0.0
36 -rwxr-xr-x 1 root root 34792 Apr 29 09:19 /usr/lib/libabsl_log_internal_check_op.so.2401.0.0
```

The compiler I'm using is GCC and I'm not specifying any generator to CMake, so it's using unix Makefiles.

carlesfernandez, 2024-05-23:
Thanks for the detailed report (and the fix!). I really need to look into this in more detail to find out why it is happening. The ```core_monitor``` lib does not use abseil, and (if I get it right) the consumer should not link against ```absl::*internal*``` libraries. Maybe the ```Debug``` build is triggering something in Protocol Buffers?

I'll be happy to make that change if it fixes the issue, but please give me some time to check what's actually happening.

castle055, 2024-05-23:
Your are right, there seems to be an issue with protobuf since version 22 where it doesn't advertise its dependencies through the CMake module.

Here is a link to the upstream issue:
https://github.com/protocolbuffers/protobuf/issues/12637

Within that thread, a comment shows a possible workaround. By using `PkgConfig` to manually find the needed dependencies and then linking to `PkgConfig::protobuf`. We would still need to call `find_package(Protobuf)` in order to use `protobuf_generate_cpp(...)` though.

Here is the link to the comment: 
https://github.com/protocolbuffers/protobuf/issues/12637#issuecomment-1871458639

I have tried it and it works, it figures out the abseil dependencies. But to me this doesn't sound like a great solution, since all targets would have to depend on both `protobuf::libprotobuf`and `PkgConfig::protobuf` targets. Otherwise, it may not work if the build system decides to build protobuf from source instead of linking to the system library.

Maybe the best solution would be to pin protobuf to an earlier version that doesn't have this issue.

carlesfernandez, 2024-05-23:
I can confirm that I can replicate the issue on Arch Linux when building in ```Debug``` mode. Interestingly, the problem does not occur when using ```Release```, ```RelWithDebInfo```, or even ```ASAN``` modes.

Additionally, if you remove the ```protobuf``` package and run ```cmake ..``` from a fresh start, CMake downloads and builds version 26.1 (the same version that is packaged), and this setup works fine in ```Debug``` mode. This indicates that the issue is likely related to the Arch Linux ```protobuf``` package.

I'm not very familiar with Arch Linux packaging, so I’m unable to provide a direct solution for that. However, a pragmatic approach might be to implement a more conservative (but still duck tape) fix. If in debug mode, if using abseil, if Protocol Buffers comes from a package, if the ```absl::log_internal_check_op``` library exists (bearing in mind that internal libraries might be removed in future versions based on Google's policies), and if whatever else we can narrow down, then link against ```absl::log_internal_check_op```. Just because it works :-( . 

I'm now triggering some jobs in our CI system to determine if this issue also occurs in other configurations that package recent versions of Abseil and Protocol Buffers. Thanks again for raising this issue and for providing the informative links.

castle055, 2024-05-23:
No worries.

Unfortunately, I cannot currently remove the `protobuf` package, as other programs depend on it. What I tried instead is to change the `find_package(Protobuf)` to `find_package(Protobuf ${GNSSSDR_PROTOCOLBUFFERS_LOCAL_VERSION})` thinking that it would then fail to find the package and build v26.1. This works but the linker error persists.

Digging through the Makefile responsible for building the specific binary I found that it is still calling `/usr/bin/protoc` in order to compile the `.proto` files. These generated files are the ones failing to link. This leads me to believe that `PROTOBUF_PROTOC_EXECUTABLE` is not being set correctly in my case where a system version of `protobuf` is available but is older than the local version.

I can try to sandbox it inside a fake root environment or a docker container tomorrow to see if that works.

Apart from that, the problem I see with linking against `absl::log_internal_check_op` is that, as you said, there is no guarantee the dependency chain will remain like that for future versions. I think the `PkgConfig` workaround might be more robust as it avoids hard coding any library names. It still needs the same conditionals to make sure it doesn't conflict with other setups.

castle055, 2024-05-24:
Closing issue as fixed by PR #774 .

Thank you @carlesfernandez for your quick responses :smiley:

closed 2024-05-24 by castle055

l29ah, 2024-10-03:
Still occurs for me at `next` branch (5be2971c9b66574fa5f086c51dccec7fea03663b):
```
[661/662] : && /usr/lib/ccache/bin/x86_64-pc-linux-gnu-g++ -O2 -pipe -march=native -ftree-vectorize -malign-data=cacheline -mtls-dialect=gnu2 -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -Wl,--hash-style=gnu src/utils/front-end-cal/CMakeFiles/front-end-cal.dir/main.cc.o -o src/utils/front-end-cal/front-end-cal  src/core/receiver/libcore_receiver.a  src/algorithms/libs/libalgorithms_libs.a  src/utils/front-end-cal/libfront_end_cal_lib.a  src/algorithms/libs/libgnss_sdr_flags.a  /usr/lib64/libabsl_flags_parse.so.2407.0.0  /usr/lib64/libabsl_log_initialize.so.2407.0.0  src/core/receiver/libcore_receiver.a  src/core/monitor/libcore_monitor.a  src/algorithms/signal_source/adapters/libsignal_source_adapters.a  src/algorithms/signal_source/gnuradio_blocks/libsignal_source_gr_blocks.a  src/algorithms/signal_source/libs/libsignal_source_libs.a  /usr/lib64/libboost_thread.so.1.85.0  /usr/lib64/libpcap.so  /usr/lib64/libpcap.so  /usr/lib64/libgnuradio-zeromq.so  /usr/lib64/libgnuradio-zeromq.so  /usr/lib64/libgnuradio-osmosdr.so  /usr/lib64/libgnuradio-osmosdr.so  src/algorithms/data_type_adapter/adapters/libdata_type_adapters.a  src/algorithms/data_type_adapter/gnuradio_blocks/libdata_type_gr_blocks.a  src/algorithms/input_filter/adapters/libinput_filter_adapters.a  src/algorithms/input_filter/gnuradio_blocks/libinput_filter_gr_blocks.a  /usr/lib64/libgnuradio-filter.so  /usr/lib64/libgnuradio-filter.so  src/algorithms/conditioner/adapters/libconditioner_adapters.a  src/algorithms/resampler/adapters/libresampler_adapters.a  src/algorithms/resampler/gnuradio_blocks/libresampler_gr_blocks.a  src/algorithms/tracking/adapters/libtracking_adapters.a  src/algorithms/tracking/gnuradio_blocks/libtracking_gr_blocks.a  src/algorithms/tracking/libs/libtracking_libs.a  src/algorithms/channel/adapters/libchannel_adapters.a  src/algorithms/telemetry_decoder/adapters/libtelemetry_decoder_adapters.a  src/algorithms/telemetry_decoder/gnuradio_blocks/libtelemetry_decoder_gr_blocks.a  src/algorithms/telemetry_decoder/libs/libswiftcnav/libtelemetry_decoder_libswiftcnav.a  src/algorithms/telemetry_decoder/libs/libtelemetry_decoder_libs.a  src/algorithms/observables/adapters/libobs_adapters.a  src/algorithms/observables/gnuradio_blocks/libobs_gr_blocks.a  src/algorithms/observables/libs/libobservables_libs.a  src/algorithms/PVT/adapters/libpvt_adapters.a  src/algorithms/PVT/gnuradio_blocks/libpvt_gr_blocks.a  src/algorithms/acquisition/adapters/libacquisition_adapters.a  src/algorithms/acquisition/gnuradio_blocks/libacquisition_gr_blocks.a  src/algorithms/acquisition/libs/libacquisition_libs.a  src/algorithms/channel/libs/libchannel_libs.a  src/core/libs/libcore_libs.a  src/core/libs/supl/libcore_libs_supl.a  src/algorithms/PVT/libs/libpvt_libs.a  src/algorithms/libs/libgnss_sdr_flags.a  /usr/lib64/libprotobuf.so  src/algorithms/libs/rtklib/libalgorithms_libs_rtklib.a  src/algorithms/libs/libalgorithms_libs.a  /usr/lib64/libgnuradio-blocks.so  /usr/lib64/libgnuradio-blocks.so  /usr/lib64/libgnuradio-fft.so  /usr/lib64/libgnuradio-fft.so  /usr/lib64/libvolk.so  /usr/lib64/libvolk.so  volk_gnsssdr_module/install/lib64/libvolk_gnsssdr.a  volk_gnsssdr_module/install/lib64/libvolk_gnsssdr.a  /usr/lib64/libarmadillo.so  /usr/lib64/libarmadillo.so  /usr/lib64/liblapack.so  /usr/lib64/libblas.so  /usr/lib64/libmatio.so  /usr/lib64/libmatio.so  /usr/lib64/libboost_system.so.1.85.0  /usr/lib64/libssl.so  /usr/lib64/libcrypto.so  /usr/lib64/libgnuradio-runtime.so  /usr/lib64/libspdlog.so.1.14.1  /usr/lib64/libfmt.so.9.1.0  src/core/system_parameters/libcore_system_parameters.a  /usr/lib64/libabsl_log_internal_conditions.so.2407.0.0  /usr/lib64/libabsl_log_internal_message.so.2407.0.0  /usr/lib64/libabsl_log_internal_log_sink_set.so.2407.0.0  /usr/lib64/libabsl_log_sink.so.2407.0.0  /usr/lib64/libabsl_log_entry.so.2407.0.0  /usr/lib64/libabsl_examine_stack.so.2407.0.0  /usr/lib64/libabsl_log_internal_format.so.2407.0.0  /usr/lib64/libabsl_log_internal_proto.so.2407.0.0  /usr/lib64/libabsl_log_internal_nullguard.so.2407.0.0  /usr/lib64/libabsl_strerror.so.2407.0.0  /usr/lib64/libboost_serialization.so.1.85.0  /usr/lib64/libpugixml.so  /usr/lib64/libpugixml.so  /usr/lib64/libboost_date_time.so.1.85.0  /usr/lib64/libgnuradio-runtime.so  /usr/lib64/libgnuradio-runtime.so  /usr/lib64/libgnuradio-pmt.so  /usr/lib64/libabsl_flags_usage.so.2407.0.0  /usr/lib64/libabsl_flags_usage_internal.so.2407.0.0  /usr/lib64/libabsl_flags_internal.so.2407.0.0  /usr/lib64/libabsl_flags_marshalling.so.2407.0.0  /usr/lib64/libabsl_flags_reflection.so.2407.0.0  /usr/lib64/libabsl_flags_config.so.2407.0.0  /usr/lib64/libabsl_cord.so.2407.0.0  /usr/lib64/libabsl_cordz_info.so.2407.0.0  /usr/lib64/libabsl_cord_internal.so.2407.0.0  /usr/lib64/libabsl_cordz_functions.so.2407.0.0  /usr/lib64/libabsl_cordz_handle.so.2407.0.0  /usr/lib64/libabsl_crc_cord_state.so.2407.0.0  /usr/lib64/libabsl_crc32c.so.2407.0.0  /usr/lib64/libabsl_str_format_internal.so.2407.0.0  /usr/lib64/libabsl_crc_internal.so.2407.0.0  /usr/lib64/libabsl_crc_cpu_detect.so.2407.0.0  /usr/lib64/libabsl_raw_hash_set.so.2407.0.0  /usr/lib64/libabsl_hashtablez_sampler.so.2407.0.0  /usr/lib64/libabsl_exponential_biased.so.2407.0.0  /usr/lib64/libabsl_flags_private_handle_accessor.so.2407.0.0  /usr/lib64/libabsl_flags_commandlineflag.so.2407.0.0  /usr/lib64/libabsl_flags_commandlineflag_internal.so.2407.0.0  /usr/lib6 [... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/core/monitor/libcore_monitor.a(gnss_synchro.pb.cc.o): undefined reference to symbol '_ZN4absl12lts_2024072212log_internal17MakeCheckOpStringIllEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEET_T0_PKc'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: /usr/lib64/libabsl_log_internal_check_op.so.2407.0.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

l29ah, 2024-10-03:
Curiously, even `-DENABLE_GLOG_AND_GFLAGS=1` doesn't prevent the error, even though `CMakeLists.txt` suggests libabsl is not required then.

l29ah, 2024-10-03:
By the way, i'm compiling a `RelWithDebInfo`.

l29ah, 2024-10-04:
Building it manually succeeds (it doesn't require the `check_op` library), yet it fails when i try to package it for Gentoo.
Full build log: https://tinystash.undef.im/il/3mBkmKsrzDtZwzCdJ8TToEKWMwa9sRDBYa3JpqnDVPWskfonLBeADumLSNABEEYnjuw6qps1KsgzktfkPqxuwY6V.log
Any ideas why it might be misbehaving?

l29ah, 2024-10-04:
Fixed it with a sledgehammer:
```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2557,14 +2557,12 @@ if(((CMAKE_CXX_COMPILER_ID STREQUAL "GNU") AND (CMAKE_CXX_COMPILER_VERSION VERSI
 else()
     find_package(Protobuf)
 
-    if((CMAKE_BUILD_TYPE STREQUAL "Debug") AND Protobuf_FOUND AND absl_FOUND)
+    if(Protobuf_FOUND AND absl_FOUND)
         # This Regular Expression is used to convert the version string provided by `find_package(Protobuf)` into the
         # appropriate binary version string. So, for instance, "4.25.3" becomes "25.3.0".
         string(REGEX REPLACE "^[0-9]+\.([0-9]+\.[0-9]+)$" "\\1.0" PROTOBUF_LIBRARY_VERSION "${Protobuf_VERSION}")
-        if((PROTOBUF_LIBRARY_VERSION VERSION_GREATER_EQUAL "22") AND (PROTOBUF_LIBRARY_VERSION VERSION_LESS "26"))
             pkg_check_modules(protobuf REQUIRED IMPORTED_TARGET protobuf=${PROTOBUF_LIBRARY_VERSION})
             target_link_libraries(protobuf::libprotobuf INTERFACE PkgConfig::protobuf)
-        endif()
     endif()
 endif()
 set_package_properties(Protobuf PROPERTIES
```

castle055, 2024-10-06:
Yep, you are correct, this solution only worked until the Protobuf version went above 26 (so outside the range). I am working on another part of the code and can't currently look into this, so I just keep bumping the upper limit version every time Protobuf gets updated. My local copy is currently at version 29. So:

```cmake
if((CMAKE_BUILD_TYPE STREQUAL "Debug") AND Protobuf_FOUND AND absl_FOUND)
    # This Regular Expression is used to convert the version string provided by `find_package(Protobuf)` into the
    # appropriate binary version string. So, for instance, "4.25.3" becomes "25.3.0".
    string(REGEX REPLACE "^[0-9]+\.([0-9]+\.[0-9]+)$" "\\1.0" PROTOBUF_LIBRARY_VERSION "${Protobuf_VERSION}")
    if((PROTOBUF_LIBRARY_VERSION VERSION_GREATER_EQUAL "22") AND (PROTOBUF_LIBRARY_VERSION VERSION_LESS "29")) # Every time the linker complains I bump this last number
        pkg_check_modules(protobuf REQUIRED IMPORTED_TARGET protobuf=${PROTOBUF_LIBRARY_VERSION})
        target_link_libraries(protobuf::libprotobuf INTERFACE PkgConfig::protobuf)
    endif()
endif()
```

We believed that the issue with Protobuf's dependencies had been solved at some point in version 26, hence the upper limit, but this seems to not be the case. So your metaphorical application of percussive maintenance may actually be the correct solution.

Also, it makes sense that building `RelWithDebInfo` would require similar linking to `Debug`. They both require debug information so the linker may link everything without optimization.

castle055, 2024-10-06:
> Building it manually succeeds (it doesn't require the `check_op` library), yet it fails when i try to package it for Gentoo. Full build log: [...]

As far as we know, this is an internal issue of Protobuf. For some reason it doesn't link properly against its own dependencies. Since it is a linker issue the code compiles just fine. It is only when linking that it fails. It makes sense that you encounter the issue while packaging since I imagine that is when the linker is called.