gnss-sdr/gnss-sdr #776: Multiple sources

SpongeBobaa, 2024-06-06:
I am trying to use USRP B210 as a receiver, with different antennas operating in the same frequency band L1, but with different RF front-end. I have set up a total of 16 channels for processing, with odd numbers for channel 0 and even numbers for channel 1. Can I achieve two different RF front ends receiving signals from the same satellite simultaneously?

My current test result shows that two RF front ends are working simultaneously, but they are receiving different satellite signals.

[image]

Thanks for your help ！

SpongeBobaa, 2024-06-07:
This is my config file

[my-gps-2.txt](https://github.com/user-attachments/files/15699308/my-gps-2.txt)

jwmelto, 2024-06-07:
What are you expecting? On the surface, this configuration doesn't make sense to me. 

If you have two different antennas, you'd want two separate instances of GNSS-SDR, since each antenna will have a different solution. 

Once a PRN is tracked, I don't believe it will be tracked again on a different channel, even from a different source.

SpongeBobaa, 2024-06-11:
I want to try GNSS-R technology on the basis of GNSS reception. One RF front-end is used to receive direct satellite signals, and the other RF front-end is used to receive signals reflected by objects from satellites.

Therefore, I raised the question in this section, which is to receive signals from the same satellite from two RF front ends.

jwmelto, 2024-06-11:
I don't believe GNSS-SDR will support what you want, out of the box. You might be able to develop an GNSS-R application using two instances of GNSS-SDR, and performing 2nd order processing on the observables each produces.

SpongeBobaa, 2024-06-11:
Okay, thank you very much for your help！

carlesfernandez, 2024-08-09:
Hi @SpongeBobaa ,

GNSS-R is a very interesting technology. Unfortunately, GNSS-SDR is not prepared for that, since it will only spend one channel per satellite, and explicitly avoids tracking the same satellite in two different channels.

closed 2024-08-09 by carlesfernandez