gnss-sdr/gnss-sdr #777: BeiDou B1I Config File

cl-cv, 2024-06-20:
I am working on generating IQ data samples for BeiDou B1I signals. I wanted to test them in GNSS-SDR, so I decided to test an existing GPS simulation I had first. After getting an output bin with GPS L1 CA samples, I ran it through GNSS-SDR with the config file shown in the image below. As expected, I got my position fix.

### GPS Config File

[image]

I then tried to convert that config file for use with B1I signals.  That file is shown below. Throughout the receiver simulation, I irregularly track some satellites and lock their secondary codes but then lose their lock shortly thereafter. Can anyone point out anything obviously wrong in this config file? I want to rule out any issues with the config before I start pointing fingers at my BeiDou signal generator.

### BDS Config File

```conf
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
;internal_fs_sps: Internal signal sampling frequency after the signal conditioning stage [samples per second].
GNSS-SDR.internal_fs_sps=5200000

;######### CONTROL_THREAD CONFIG ############
ControlThread.wait_for_flowgraph=false

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/testuser/gnss-sdr/conf/File_input/Beidou/BDS_Testing/bdssim.bin
SignalSource.item_type=ishort
SignalSource.sampling_frequency=5200000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

DataTypeAdapter.implementation=Ishort_To_Complex
DataTypeAdapter.dump=false

InputFilter.implementation=Pass_Through
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex

Resampler.implementation=Pass_Through
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_B1.count=5
Channels.in_acquisition=1
Channel.signal=B1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
Acquisition_B1.item_type=gr_complex
Acquisition_B1.coherent_integration_time_ms=1
Acquisition_B1.pfa=0.01
Acquisition_B1.doppler_max=10000
Acquisition_B1.doppler_step=100
Acquisition_B1.dump=false
Acquisition_B1.dump_filename=./bds_acq

;######### TRACKING GLOBAL CONFIG ############
Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
Tracking_B1.item_type=gr_complex
Tracking_B1.pll_bw_hz=25.0;
Tracking_B1.dll_bw_hz=2.5;
Tracking_B1.order=3;
Tracking_B1.dump=false;
Tracking_B1.dump_filename=./epl_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_B1.implementation=BEIDOU_B1I_Telemetry_Decoder
TelemetryDecoder_B1.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=PPP_Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.dump_filename=./PVT
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=false
```

closed 2024-07-01 by cl-cv

cehuidong, 2024-07-18:
Hi. 
I'm doing something similar to yours, I'm using a similar method to generate the Beidou B1I signal, but I'm having the same problem with the B1I signal being loss their lock. I can't really whether it's a signal generation error or a GNSS-SDR configuration issue. May I ask if you have solved it now?

cl-cv, 2024-07-18:
> Hi. I'm doing something similar to yours, I'm using a similar method to generate the Beidou B1I signal, but I'm having the same problem with the B1I signal being loss their lock. I can't really [...]

Hi,
Yes, I have mine figured out now, although my issue was primarily in signal generation. Nonetheless, here is my working config file for reference

```conf
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
;internal_fs_sps: Internal signal sampling frequency after the signal conditioning stage [samples per second].
GNSS-SDR.internal_fs_sps=5200000
GNSS-SDR.telecommand_enabled=true
GNSS-SDR.telecommand_tcp_port=3333

;######### CONTROL_THREAD CONFIG ############
ControlThread.wait_for_flowgraph=false

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=PATH_HERE
SignalSource.item_type=ishort
SignalSource.sampling_frequency=5200000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=true
SignalSource.dump_filename=./signalsource/source.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex

InputFilter.implementation=Pass_Through
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex

Resampler.implementation=Pass_Through
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_B1.count=12
Channels.in_acquisition=1
Channel.signal=B1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
Acquisition_B1.item_type=gr_complex
Acquisition_B1.coherent_integration_time_ms=1
Acquisition_B1.pfa=0.02
Acquisition_B1.max_dwells=2
;Acquisition_B1.pfa=0.0000001;
Acquisition_B1.doppler_max=5000
Acquisition_B1.doppler_min=-5000
Acquisition_B1.doppler_step=50
Acquisition_B1.dump=true
Acquisition_B1.dump_filename=./acquisition/bds_acq
Acquisition_B1.blocking=true;
Acquisition_B1.bit_transition_flag = false;

;######### TRACKING GLOBAL CONFIG ############
Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
Tracking_B1.item_type=gr_complex
Tracking_B1.pll_bw_hz=25.0;
Tracking_B1.dll_bw_hz=2.50;
Tracking_B1.extend_correlation_symbols=10
Tracking_B1.cn0_min=27
;Tracking_B1.cn0_min=1
Tracking_B1.dump=true
Tracking_B1.dump_filename=./tracking/epl_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_B1.implementation=BEIDOU_B1I_Telemetry_Decoder
TelemetryDecoder_B1.dump=true
TelemetryDecoder_B1.dump_filename=./telemetry/telemetry.dat

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_filename=./observables/observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.dump_filename=./PVT
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=true
PVT.rinex_version=3
PVT.rinex_output_enabled=true
PVT.gpx_output_enabled=true
PVT.raim_fde=0
PVT.enable_monitor=true
PVT.enable_protobuf=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1111

Monitor.enable_monitor=true
Monitor.decimation_factor=1
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1112
```

cehuidong, 2024-07-19:
> > 你好。我正在做与您类似的事情，我正在使用类似的方法来生成北斗 B1I 信号，但我遇到了同样的问题，即 B1I 信号丢失了锁定。我真的无法确定是信号生成错误还是 GNSS-SDR 配置问题。请问你现在是否已经解决了？
> 
> 嗨，是的，我现在已经想通了，尽管我的问题主要出在信号生成上。尽管如此，这是我的工作配置文件供参考
> 
> ```
> [GNSS-SDR]
> 
> ;######### GLOBAL OPTIONS ##################
> ;internal_fs_sps: Internal signal sampling frequency after the signal conditioning stage [samples per second].
> GNSS-SDR.internal_fs_sps=5200000
> GNSS-SDR.telecommand_enabled=true
> GNSS-SDR.telecommand_tcp_port=3333
> 
> ;######### CONTROL_THREAD CONFIG ############
> ControlThread.wait_for_flowgraph=false
> 
> ;######### SIGNAL_SOURCE CONFIG ############
> SignalSource.implementation=File_Signal_Source
> SignalSource.filename=PATH_HERE
> SignalSource.item_type=ishort
> SignalSource.sampling_frequency=5200000
> SignalSource.samples=0
> SignalSource.repeat=false
> SignalSource.dump=true
> SignalSource.dump_filename=./signalsource/source.dat
> SignalSource.enable_throttle_control=false
> 
> ;######### SIGNAL_CONDITIONER CONFIG ############
> SignalConditioner.implementation=Signal_Conditioner
> DataTypeAdapter.implementation=Ishort_To_Complex
> 
> InputFilter.implementation=Pass_Through
> InputFilter.input_item_type=gr_complex
> InputFilter.output_item_type=gr_complex
> 
> Resampler.implementation=Pass_Through
> Resampler.item_type=gr_complex
> 
> 
> ;######### CHANNELS GLOBAL CONFIG ############
> Channels_B1.count=12
> Channels.in_acquisition=1
> Channel.signal=B1
> 
> 
> ;######### ACQUISITION GLOBAL CONFIG ############
> Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
> Acquisition_B1.item_type=gr_complex
> Acquisition_B1.coherent_integration_time_ms=1
> Acquisition_B1.pfa=0.02
> Acquisition_B1.max_dwells=2
> ;Acquisition_B1.pfa=0.0000001;
> Acquisition_B1.doppler_max=5000
> Acquisition_B1.doppler_min=-5000
> Acquisition_B1.doppler_step=50
> Acquisition_B1.dump=true
> Acquisition_B1.dump_filename=./acquisition/bds_acq
> Acquisition_B1.blocking=true;
> Acquisition_B1.bit_transition_flag = false;
> 
> 
> ;######### TRACKING GLOBAL CONFIG ############
> Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
> Tracking_B1.item_type=gr_complex
> Tracking_B1.pll_bw_hz=25.0;
> Tracking_B1.dll_bw_hz=2.50;
> Tracking_B1.extend_correlation_symbols=10
> Tracking_B1.cn0_min=27
> ;Tracking_B1.cn0_min=1
> Tracking_B1.dump=true
> Tracking_B1.dump_filename=./tracking/epl_tracking_ch_
> 
> 
> ;######### TELEMETRY DECODER GPS CONFIG ############
> TelemetryDecoder_B1.implementation=BEIDOU_B1I_Telemetry_Decoder
> TelemetryDecoder_B1.dump=true
> TelemetryDecoder_B1.dump_filename=./telemetry/telemetry.dat
> 
> 
> ;######### OBSERVABLES CONFIG ############
> Observables.implementation=Hybrid_Observables
> Observables.dump=true
> Observables.dump_filename=./observables/observables.dat
> 
> 
> ;######### PVT CONFIG ############
> PVT.implementation=RTKLIB_PVT
> PVT.positioning_mode=Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
> PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
> PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
> PVT.output_rate_ms=100
> PVT.display_rate_ms=500
> PVT.dump_filename=./PVT
> PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
> PVT.flag_nmea_tty_port=false;
> PVT.nmea_dump_devname=/dev/pts/4
> PVT.flag_rtcm_server=false
> PVT.flag_rtcm_tty_port=false
> PVT.rtcm_dump_devname=/dev/pts/1
> PVT.dump=true
> PVT.rinex_version=3
> PVT.rinex_output_enabled=true
> PVT.gpx_output_enabled=true
> PVT.raim_fde=0
> PVT.enable_monitor=true
> PVT.enable_protobuf=true
> PVT.monitor_client_addresses=127.0.0.1
> PVT.monitor_udp_port=1111
> 
> Monitor.enable_monitor=true
> Monitor.decimation_factor=1
> Monitor.client_addresses=127.0.0.1
> Monitor.udp_port=1112
> ```

Thank you very much for your help. Unfortunately, I still haven't solved the problem and it looks like I have a serious error in the signal generation. I have some questions on generating BeiDou B1I signals:
1、I used the same signal modulation as gps-sdr-sim:
ip = chan[i].dataBit * chan[i].codeCA * cosTable512[iTable] * gain[i];
qp = chan[i].dataBit * chan[i].codeCA * sinTable512[iTable] * gain[i];
Is this modulation scheme correct for BeiDou B1I, and are there specific synchronization considerations I should be aware of?
2、I have generated the D1 navigation message following the gps-sdr-sim method, including:
Subframes 1, 2, and 3
Page 1 of subframe 4
Page 7 of subframe 5
Is this sufficient for generating a functional BeiDou B1I signal, or are there additional subframes/pages I should include?