gnss-sdr/gnss-sdr #779: Try "My first position fix Configurations" but got position time is 2032-Nov-18

SnowWhite0808, 2024-06-25:
Hello, I am a new beginner learning GNSS and want to use GNSS-SDR, I installed dependencies on my laptop following [gnss-sdr](https://github.com/gnss-sdr/gnss-sdr/tree/main)
/README.md, which is macOS, then I installed GNSS-SDR by this command: sudo port install gnss-sdr, then I followed "My first position fix Configurations"

I checked the GNSS-SDR version and got: gnss-sdr version 0.0.19
Debug build (NDEBUG not #defined), but I have no idea why it is built in debug model.

Then I downloaded the source data and copied the config file, but the position I got is on 2032-Nov-18, like the following:

Current receiver time: 1 min 36 s
Position at 2032-Nov-18 06:24:40.000000 UTC using 5 observations is Lat = 41.274851 [deg], Long = 1.987711 [deg], Height = 72.67 [m]
Velocity: East: -0.25 [m/s], North: -0.15 [m/s], Up = 0.27 [m/s]
Position at 2032-Nov-18 06:24:40.500000 UTC using 5 observations is Lat = 41.274858 [deg], Long = 1.987737 [deg], Height = 71.41 [m]
Velocity: East: -0.79 [m/s], North: -1.43 [m/s], Up = 0.88 [m/s]
Current receiver time: 1 min 37 s
Position at 2032-Nov-18 06:24:41.000000 UTC using 5 observations is Lat = 41.274851 [deg], Long = 1.987709 [deg], Height = 74.07 [m]
Velocity: East: 0.61 [m/s], North: -0.95 [m/s], Up = -0.76 [m/s]
Position at 2032-Nov-18 06:24:41.500000 UTC using 5 observations is Lat = 41.274811 [deg], Long = 1.987687 [deg], Height = 73.57 [m]
Velocity: East: 0.98 [m/s], North: 0.09 [m/s], Up = -0.22 [m/s]
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 08 (Block IIF)
Current receiver time: 1 min 38 s
New GPS NAV message received in channel 6: subframe 5 from satellite GPS PRN 20 (Block IIR) with CN0=44.25 dB-Hz
New GPS NAV message received in channel 0: subframe 5 from satellite GPS PRN 01 (Block IIF) with CN0=47.92 dB-Hz
New GPS NAV message received in channel 4: subframe 5 from satellite GPS PRN 32 (Block IIF) with CN0=44.97 dB-Hz
New GPS NAV message received in channel 7: subframe 5 from satellite GPS PRN 17 (Block IIR-M) with CN0=45.03 dB-Hz
New GPS NAV message received in channel 3: subframe 5 from satellite GPS PRN 11 (Block III) with CN0=45.95 dB-Hz
Loss of lock in channel 1!
Position at 2032-Nov-18 06:24:42.000000 UTC using 5 observations is Lat = 41.274854 [deg], Long = 1.987693 [deg], Height = 75.17 [m]
Velocity: East: 0.46 [m/s], North: 0.62 [m/s], Up = -0.39 [m/s]
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 21 (Block IIR)
Position at 2032-Nov-18 06:24:42.500000 UTC using 5 observations is Lat = 41.274858 [deg], Long = 1.987703 [deg], Height = 82.94 [m]
Velocity: East: -0.28 [m/s], North: 0.72 [m/s], Up = 1.02 [m/s]
Current receiver time: 1 min 39 s
Position at 2032-Nov-18 06:24:43.000000 UTC using 5 observations is Lat = 41.274860 [deg], Long = 1.987698 [deg], Height = 69.69 [m]
Velocity: East: -0.49 [m/s], North: -0.56 [m/s], Up = -0.54 [m/s]
Position at 2032-Nov-18 06:24:43.500000 UTC using 5 observations is Lat = 41.274839 [deg], Long = 1.987690 [deg], Height = 79.33 [m]
Velocity: East: 0.52 [m/s], North: 0.61 [m/s], Up = -2.04 [m/s]
Current receiver time: 1 min 40 s
Position at 2032-Nov-18 06:24:44.000000 UTC using 5 observations is Lat = 41.274812 [deg], Long = 1.987626 [deg], Height = 61.89 [m]
Velocity: East: 0.08 [m/s], North: -0.29 [m/s], Up = 1.22 [m/s]
Position at 2032-Nov-18 06:24:44.500000 UTC using 5 observations is Lat = 41.274865 [deg], Long = 1.987682 [deg], Height = 70.20 [m]
Velocity: East: -0.22 [m/s], North: -0.36 [m/s], Up = -0.72 [m/s]
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 6.897808 [seconds]
GNSS-SDR program ended.

Does anyone know the reason? How can I fix it? 
Any suggestions are welcome, thanks in advance!

SnowWhite0808, 2024-06-25:
This is my config file: 

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=./work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat
SignalSource.item_type=ishort
SignalSource.sampling_frequency=4000000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=4000000
Resampler.sample_freq_out=2000000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=8
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250
Acquisition_1C.blocking=true

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

jwmelto, 2024-06-26:
I think this is normal. GPS only reports time as week number and seconds of week. The week number rolls over every 1024 weeks or so. There has been a rollover since that file was recorded. GNSS-SDR is biased towards real time processing, so it assumes the week number is in the current epoch. So it makes sense that your times are in the future. 

I haven’t run that file lately but you can look up the epoch rollovers and do the math for yourself.

SnowWhite0808, 2024-06-26:
> I think this is normal. GPS only reports time as week number and seconds of week. The week number rolls over every 1024 weeks or so. There has been a rollover since that file was recorded. GNSS-SDR [...]

Yes, that makes sense. I changed my laptop date to be before 2019 and the result shows 4/4/2013. Thanks a lot!!
Besides, I found there are some "Loss of lock in channel 4!"  showing in the results:

Current receiver time: 1 min 22 s
Loss of lock in channel 4!
Position at 2013-Apr-04 06:24:28.000000 UTC using 5 observations is Lat = 41.274865 [deg], Long = 1.987715 [deg], Height = 74.43 [m]
Velocity: East: 1.28 [m/s], North: 1.93 [m/s], Up = -1.61 [m/s]
Position at 2013-Apr-04 06:24:28.500000 UTC using 5 observations is Lat = 41.274843 [deg], Long = 1.987703 [deg], Height = 62.69 [m]
Velocity: East: -0.27 [m/s], North: 0.43 [m/s], Up = 0.07 [m/s]
Current receiver time: 1 min 23 s
Loss of lock in channel 2!
Position at 2013-Apr-04 06:24:29.000000 UTC using 5 observations is Lat = 41.274820 [deg], Long = 1.987687 [deg], Height = 69.34 [m]
Velocity: East: -0.47 [m/s], North: -1.56 [m/s], Up = 1.71 [m/s]
Position at 2013-Apr-04 06:24:29.500000 UTC using 5 observations is Lat = 41.274862 [deg], Long = 1.987700 [deg], Height = 76.90 [m]
Velocity: East: 0.34 [m/s], North: 0.72 [m/s], Up = -0.58 [m/s]
Current receiver time: 1 min 24 s
Position at 2013-Apr-04 06:24:30.000000 UTC using 5 observations is Lat = 41.274795 [deg], Long = 1.987668 [deg], Height = 75.46 [m]
Velocity: East: 0.09 [m/s], North: 0.29 [m/s], Up = -5.11 [m/s]
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 24 (Block IIF)
Position at 2013-Apr-04 06:24:30.500000 UTC using 5 observations is Lat = 41.274867 [deg], Long = 1.987685 [deg], Height = 70.38 [m]
Velocity: East: 0.28 [m/s], North: -0.60 [m/s], Up = 1.71 [m/s]

Also, I tried "volk_gnsssdr_profile" in the terminal, but the result shows"illegal hardware instruction  volk_gnsssdr_profile"    
the following is what I got:                                  
RUN_VOLK_GNSSSDR_TESTS: volk_gnsssdr_8i_accumulator_s8i(8111,100000)
generic completed in 341.253 ms
zsh: illegal hardware instruction  volk_gnsssdr_profile

I searched on the Internet, and it seems sth is not compatible with my CPU? I am not sure.

Any ideas about these two issues? 

Thanks. I appreciate your help.

carlesfernandez, 2024-07-01:
Ups, we have been bitten by the "mid rollover"!

This can be solved by adding this line to the configuration file:

```
GNSS-SDR.pre_2009_file=true
```

... but this should be renamed to `pre_2019`.

jwmelto, 2024-07-01:
We probably need either another flag, or a revised expression. Possibly a "year of collection"?

XueWei08, 2024-07-01:
> Ups, we have been bitten by the "mid rollover"!
> 
> This can be solved by adding this line to the configuration file:
> 
> ```
> GNSS-SDR.pre_2009_file=true
> ```
> 
> ... but this should be renamed to `pre_2019`.

Thanks! it works.

Also, I tried "volk_gnsssdr_profile" in the terminal, but the result shows"illegal hardware instruction volk_gnsssdr_profile"
the following is what I got:
RUN_VOLK_GNSSSDR_TESTS: volk_gnsssdr_8i_accumulator_s8i(8111,100000)
generic completed in 341.253 ms
zsh: illegal hardware instruction volk_gnsssdr_profile

I searched on the Internet, and it seems sth is not compatible with my CPU? I am not sure.

And there are a lot of "Loss of lock in channel "  in the results. Is it normal?

Any ideas about these two issues?

jwmelto, 2024-07-02:
Did you build from source or install a pre-built package? If package, which one? What CPU do you have? Yes, it seems something is not compatible with your CPU but unless you give more details, no one else can help.

Loss of lock frequently is associated with dropped data, but for a file source, that should not be possible. However, there is some time-dependent interaction between the acquisition and tracking blocks; you might try setting `Signal_Source.enable_throttle_control=true`; this will slow you down to real-time, but might help your stability.

The other reason for loss of lock is if the signal truly drops below the detection threshold in the noise floor, as can happen when the satellite sets. I'm not familiar enough with this file to know what "normal" is.

XueWei08, 2024-07-02:
> Did you build from source or install a pre-built package? If package, which one? What CPU do you have? Yes, it seems something is not compatible with your CPU but unless you give more details, no [...]

Hi, thanks for your replying. I built it from source and cloned from "next" branch, and  I am using MacBook Pro 14(2021), the chip is Apple M1 Pro, macOS Sonoma Version 14.5

I am following this example: https://gnss-sdr.org/my-first-fix/   and downloaded the file like this:
[image]

I added that comment line in the conf file, but still see some ”loss“, I copied the whole result here:

gnss-sdr --config_file=./my_first_gnss-sdr_receiver.conf 
Initializing GNSS-SDR v0.0.19.git-next-d62e2e0de ... Please wait.
Logging will be written at "/var/folders/q1/6178sx112r94ldkmjrvkqgvh0000gn/T/"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Processing file ./work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 800000000 samples (1600000000 bytes)
GNSS signal recorded time to be processed: 99.95 [s]
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
   2048 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 16384 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   431 items of size 152. Due to alignment requirements
[... 484 lines cut ...]
Velocity: East: 0.98 [m/s], North: 0.09 [m/s], Up = -0.23 [m/s]
Current receiver time: 1 min 38 s
New GPS NAV message received in channel 5: subframe 5 from satellite GPS PRN 20 (Block IIR) with CN0=44.25 dB-Hz
New GPS NAV message received in channel 0: subframe 5 from satellite GPS PRN 01 (Block IIF) with CN0=47.92 dB-Hz
New GPS NAV message received in channel 4: subframe 5 from satellite GPS PRN 32 (Block IIF) with CN0=44.97 dB-Hz
New GPS NAV message received in channel 7: subframe 5 from satellite GPS PRN 11 (Block III) with CN0=45.95 dB-Hz
New GPS NAV message received in channel 6: subframe 5 from satellite GPS PRN 17 (Block IIR-M) with CN0=45.03 dB-Hz
Position at 2013-Apr-04 06:24:44.000000 UTC using 5 observations is Lat = 41.274853 [deg], Long = 1.987693 [deg], Height = 75.14 [m]
Velocity: East: 0.46 [m/s], North: 0.62 [m/s], Up = -0.39 [m/s]
Position at 2013-Apr-04 06:24:44.500000 UTC using 5 observations is Lat = 41.274858 [deg], Long = 1.987703 [deg], Height = 82.93 [m]
Velocity: East: -0.28 [m/s], North: 0.72 [m/s], Up = 1.02 [m/s]
Current receiver time: 1 min 39 s
Position at 2013-Apr-04 06:24:45.000000 UTC using 5 observations is Lat = 41.274859 [deg], Long = 1.987698 [deg], Height = 69.69 [m]
Velocity: East: -0.49 [m/s], North: -0.56 [m/s], Up = -0.53 [m/s]
[... cut ...]
Position at 2013-Apr-04 06:24:46.000000 UTC using 5 observations is Lat = 41.274812 [deg], Long = 1.987626 [deg], Height = 61.89 [m]
Velocity: East: 0.08 [m/s], North: -0.29 [m/s], Up = 1.22 [m/s]
Position at 2013-Apr-04 06:24:46.500000 UTC using 5 observations is Lat = 41.274865 [deg], Long = 1.987682 [deg], Height = 70.18 [m]
Velocity: East: -0.22 [m/s], North: -0.37 [m/s], Up = -0.73 [m/s]
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 7.407247 [seconds]
GNSS-SDR program ended.

jwmelto, 2024-07-02:
> RUN_VOLK_GNSSSDR_TESTS: volk_gnsssdr_8i_accumulator_s8i(8111,100000)
generic completed in 341.253 ms
zsh: illegal hardware instruction volk_gnsssdr_profile

This appears to be a compatibility issue. It is not directly related to your performance, because the default kernels should be fine. I don't know if any effort has been made to tune the GNSSSDR VOLK kernels for M1 chips.

> Total GNSS-SDR run time: 7.407247 [seconds]

This doesn't looks like the throttle had the desired effect. There are nearly 100 seconds of data that you processed in 7 seconds. I see there was a typo in my previous comment; the proper prefix is `SignalSource`. I've updated the comment for posterity.

On the whole, you are getting a position solution; it is substantively consistent and correct. I wouldn't worry too much about transient cruft.

closed 2024-07-09 by SnowWhite0808