gnss-sdr/gnss-sdr #780: Proper Use of ZMQ_Signal_Source

wmszeliga, 2024-06-30:
I'm looking for some clarification on what the expected message format is for a ZMQ_Signal_Source. I have a [GNSS Firehose](https://transitiva.com/product/gnss-firehose-receiver-tri-band-quad-constellation/) and have written an [adapter](https://github.com/wmszeliga/pcap_zmq) that parses the ethernet packets from the Firehose and publishes them to a ZMQ PUBSUB. Right now, each message is 960 signed byte (int8_t) interleaved I/Q samples for a total message size of 1920 bytes. If I use a ZMQ_Signal_Source with vlen=1920 and item_type=ibyte, tracking begins for visible satellites, but quickly looses lock. Creating a constellation diagram of prompt I and prompt Q from the tracking loop shows one cloud of I/Q points clustered around the origin, so I'm essentially not seeing the BPSK signal.

Here's the interesting part; If I subscribe to the same PUBSUB with a super-simple Python script and just dump the raw I/Q samples to a file and then process that file with GNSS-SDR using the File_Signal_Source with item_type=ibyte, the tracking loop works just fine and the constellation diagram shows two clouds offset from the origin along the x-axis.

Setting the SignalSource.dump=true in the ZMQ_Signal_Source while simultaneously capturing and dumping messages using the Python script (and Ctrl-c'ing them at the same time) should yield file dumps of nearly the same size (one from GNSS-SDR's SignalSource=ZMQ_Signal_Source and one from the separate Python script subscriber), however the file size for the Python dump is approximately 400% larger.

One additional thing I did try was serializing the data with PMT before putting it in the ZMQ message, but that yields a message that is slightly larger than 1920 bytes and causes confusion in the GNSS-SDR SignalSource block, so I never technically got that working. Does anyone have any advice?

jwmelto, 2024-07-02:
Simply put, the ZMQ_Signal_Source is designed to consume a GNU Radio [ZMQ Pub Sink](https://wiki.gnuradio.org/index.php/ZMQ_PUB_Sink). Trying to marshal this data directly is challenging. There's a little information [here](https://wiki.gnuradio.org/index.php?title=Understanding_ZMQ_Blocks).

Also, bear in mind that the ZMQ blocks are _vector_ blocks. When I use this, I have a source that sends 1024 samples per message.

I have used the block to allow GNSS-SDR to respond either to a radio front-end (simple adapter that feeds an ZMQ Pub Sink) or a file source (custom chain that runs file source->throttle->vector->ZMQ Pub Sink).

wmszeliga, 2024-07-02:
@jwmelto Can you share the example for the radio front-end? There is no front-end for the Firehose, so this was my first attempt at developing a way to get the data in. I have also tried FIFO pipes, but that approach really ends up with quite a bit of backpressure and I wasn't able to get the data stream to work.

jwmelto, 2024-07-02:
Not easily, but I'll try to figure it out. Once you've written your first GNU Radio block, it gets easier. Give me a few days.

jwmelto, 2024-07-02:
Actually, my front end was a CyberRadioSolutions box and I used their driver. I won't (necessarily) recommend the implementation, but you can look at it [here](https://github.com/CyberRadio/gr-cyberradio/blob/c8800e82b95015a479d17c31b60f81992e2b1006/gr-CyberRadio/lib/vita_udp_rx_impl.cc#L582) for reference. I just hooked that block up to a pub sink.

wmszeliga, 2024-07-02:
Thanks. It sounds as if I need to write a driver for the Firehose and GnuRadio. I'll start looking more closely at the GNU Radio drivers as an example.

jwmelto, 2024-07-03:
Here is a simple app I wrote to read a file and publish it to ZMQ. The corresponding SignalSource to consume it is
```
SignalSource.implementation=ZMQ_Signal_Source
SignalSource.endpoint=tcp://127.0.0.1:16192    ;# whatever endpoint you choose
SignalSource.vlen=1024                         ;# must match sending vector length
SignalSource.pass_tags=true                    ;# must match sending block
SignalSource.dump=false
SignalSource.enable_stats=true
```

[file_to_ZMQ.py.txt](https://github.com/user-attachments/files/16086837/file_to_ZMQ.py.txt)

jmfriedt, 2024-08-04:
If it can be of any use, here is my approach for sending data collected by a PocketSDR to gnss-sdr: no need to write a GNU Radio dedicated source, named pipe FIFO from PocketSDR pocket_dump source to GNU Radio input, GNU Radio handles frequency transposition and decimation (could probably be done by gnss-sdr but was a nice example of streaming through GNU Radio), and output to 0MQ.

[image]

See https://github.com/jmfriedt/max2771_fx2lp/tree/main/240729_stability for details.

jmfriedt, 2024-08-04:
> Also, bear in mind that the ZMQ blocks are _vector_ blocks. When I use this, I have a source that sends 1024 samples per message.

Just checked the documentation, https://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-zmq_signal_source states that the default vlen size is 1.

closed 2024-09-20 by carlesfernandez