gnss-sdr/gnss-sdr #783: several "Loss of lock in channel !" and No NAV message received in real-time GNSS-SDR

XueWei08, 2024-07-03:
Hello, I am trying to use GNSS-SDR on real-time signals, I built GNSS-SDR from source and cloned from "next" branch, and I am using MacBook Pro 14(2021), the chip is Apple M1 Pro, macOS Sonoma Version 14.5, the antenna I am using is Tallysman Triple Band GNSS Antenna 33-7972-07, and I feed the antenna with 12V power and a bias-tee ZFBT-352-FT+, and I am using USRP B200, the conf file I am using is :

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000
GNSS-SDR.pre_2009_file=true
GNSS-SDR.pre_2019_file=true

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.device_address=
;  <- PUT THE IP ADDRESS OF YOUR USRP HERE OR LEAVE IT EMPTY FOR USB
SignalSource.item_type=cshort
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
SignalSource.gain=32
SignalSource.subdevice=A:A  ; <- Can be A:0 or B:0
SignalSource.samples=0
SignalSource.enable_throttle_control=true

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=cshort

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Fir_Filter
InputFilter.input_item_type=cshort
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=11
InputFilter.number_of_bands=2

InputFilter.band1_begin=0.0
InputFilter.band1_end=0.48
InputFilter.band2_begin=0.52
InputFilter.band2_end=1.0

InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0

InputFilter.band1_error=1.0
InputFilter.band2_error=1.0

InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=4000000
InputFilter.IF=0

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
[... 349 lines cut ...]
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 18 (Block III)
Loss of lock in channel 3!
Current receiver time: 1 min 59 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 05 (Block IIR-M)
Current receiver time: 2 min 0 s
Current receiver time: 2 min 1 s
Current receiver time: 2 min 2 s
Current receiver time: 2 min 3 s
Loss of lock in channel 5!
Current receiver time: 2 min 4 s
Loss of lock in channel 2!
Loss of lock in channel 1!
Current receiver time: 2 min 5 s
Current receiver time: 2 min 6 s
Current receiver time: 2 min 7 s
Current receiver time: 2 min 8 s
Current receiver time: 2 min 9 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 2 min 10 s
Current receiver time: 2 min 11 s
OTracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 09 (Block IIF)
OCurrent receiver time: 2 min 12 s

Does anyone know the reason? How can I fix it?
Any suggestions are welcome, thanks in advance!

jwmelto, 2024-07-04:
As noted [here](https://github.com/gnss-sdr/gnss-sdr/issues/779#issuecomment-2201650173), loss of lock is frequently related to dropped data. You are definitely dropping data.

> Loss of lock in channel 0!
Current receiver time: 22 s
Current receiver time: 23 s
OLoss of lock in channel 4!
Current receiver time: 24 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 30 (Block IIF)
Current receiver time: 25 s
Current receiver time: 26 s
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 08 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 06 (Block IIF)
Current receiver time: 27 s
Current receiver time: 28 s
OCurrent receiver time: 29 s
Loss of lock in channel 5!

Those 'O' at the beginning of a line come from the radio driver and indicate an "overflow" condition, meaning you aren't reading the data fast enough to keep up with it.

You said you built from source; is this a debug build or a release build? You very well may need the optimization of a release build to handle the data rate.

You can also try to reduce the sample rate; GPS only needs about 2msps sample rate to process adequately.

Make sure you are on a USB-2.0 port, and whatever hub you are connected through is giving you full bandwidth.

XueWei08, 2024-07-05:
> As noted [here](https://github.com/gnss-sdr/gnss-sdr/issues/779#issuecomment-2201650173), loss of lock is frequently related to dropped data. You are definitely dropping data.
>
>
> Those 'O' at the [...]

Hi, I am using the release version, and I modified the sampling rate and tried again, I use still USB-3.0 port because I don't have USB2.0 with me now,  I got the position message, the Lat and Long are correct, and the height is about 100 m but I put the antenna on the ground, so I am expecting the height should not be that high. Is this because the height is a related value or some other reason related to the position calculation?

This is the result I got:

Loss of lock in channel 6!
Current receiver time: 1 min 11 s
Position at 2024-Jul-04 03:17:23.000000 UTC using 4 observations is Lat = 39.154847 [deg], Long = -77.190734 [deg], Height = 103.05 [m]
Velocity: East: -0.03 [m/s], North: 0.01 [m/s], Up = 0.05 [m/s]
Position at 2024-Jul-04 03:17:23.500000 UTC using 4 observations is Lat = 39.154888 [deg], Long = -77.190754 [deg], Height = 104.31 [m]
Velocity: East: 0.02 [m/s], North: 0.01 [m/s], Up = 0.10 [m/s]
New GPS NAV message received in channel 0: subframe 2 from satellite GPS PRN 01 (Block IIF) with CN0=43.27 dB-Hz
Satellite GPS PRN 01 (Block IIF) reports an unhealthy status, not used for navigation
New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 26 (Block IIF) with CN0=44.87 dB-Hz
New GPS NAV message received in channel 3: subframe 2 from satellite GPS PRN 31 (Block IIR-M) with CN0=43.09 dB-Hz
New GPS NAV message received in channel 1: subframe 2 from satellite GPS PRN 03 (Block IIF) with CN0=41.90 dB-Hz
New GPS NAV message received in channel 2: subframe 2 from satellite GPS PRN 28 (Block IIR) with CN0=43.09 dB-Hz
Current receiver time: 1 min 12 s
Position at 2024-Jul-04 03:17:24.000000 UTC using 4 observations is Lat = 39.154922 [deg], Long = -77.190715 [deg], Height = 113.01 [m]
Velocity: East: 0.02 [m/s], North: 0.03 [m/s], Up = -0.13 [m/s]
Position at 2024-Jul-04 03:17:24.500000 UTC using 4 observations is Lat = 39.154892 [deg], Long = -77.190710 [deg], Height = 99.93 [m]
Velocity: East: 0.00 [m/s], North: -0.02 [m/s], Up = -0.05 [m/s]
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 05 (Block IIR-M)
Current receiver time: 1 min 13 s
Position at 2024-Jul-04 03:17:25.000000 UTC using 4 observations is Lat = 39.154908 [deg], Long = -77.190742 [deg], Height = 103.71 [m]
Velocity: East: -0.02 [m/s], North: -0.03 [m/s], Up = -0.03 [m/s]
Position at 2024-Jul-04 03:17:25.500000 UTC using 4 observations is Lat = 39.154924 [deg], Long = -77.190757 [deg], Height = 108.51 [m]
Velocity: East: 0.01 [m/s], North: -0.01 [m/s], Up = -0.02 [m/s]
Current receiver time: 1 min 14 s
Position at 2024-Jul-04 03:17:26.000000 UTC using 4 observations is Lat = 39.154854 [deg], Long = -77.190741 [deg], Height = 103.98 [m]
Velocity: East: 0.02 [m/s], North: -0.00 [m/s], Up = -0.06 [m/s]
Position at 2024-Jul-04 03:17:26.500000 UTC using 4 observations is Lat = 39.154857 [deg], Long = -77.190754 [deg], Height = 113.29 [m]
Velocity: East: -0.01 [m/s], North: -0.00 [m/s], Up = 0.08 [m/s]
Current receiver time: 1 min 15 s
Position at 2024-Jul-04 03:17:27.000000 UTC using 4 observations is Lat = 39.154889 [deg], Long = -77.190745 [deg], Height = 109.17 [m]
Velocity: East: -0.02 [m/s], North: -0.00 [m/s], Up = -0.01 [m/s]
Position at 2024-Jul-04 03:17:27.500000 UTC using 4 observations is Lat = 39.154907 [deg], Long = -77.190754 [deg], Height = 117.16 [m]
Velocity: East: -0.01 [m/s], North: -0.01 [m/s], Up = -0.07 [m/s]
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 10 (Block IIF)
Current receiver time: 1 min 16 s
Position at 2024-Jul-04 03:17:28.000000 UTC using 4 observations is Lat = 39.154893 [deg], Long = -77.190750 [deg], Height = 113.47 [m]
Velocity: East: -0.00 [m/s], North: 0.12 [m/s], Up = 0.06 [m/s]
Position at 2024-Jul-04 03:17:28.500000 UTC using 4 observations is Lat = 39.154905 [deg], Long = -77.190741 [deg], Height = 111.33 [m]
Velocity: East: -0.00 [m/s], North: -0.07 [m/s], Up = -0.12 [m/s]
Current receiver time: 1 min 17 s
Position at 2024-Jul-04 03:17:29.000000 UTC using 4 observations is Lat = 39.154868 [deg], Long = -77.190692 [deg], Height = 107.06 [m]
Velocity: East: 0.03 [m/s], North: -0.03 [m/s], Up = -0.06 [m/s]
Position at 2024-Jul-04 03:17:29.500000 UTC using 4 observations is Lat = 39.154891 [deg], Long = -77.190691 [deg], Height = 105.51 [m]
Velocity: East: -0.01 [m/s], North: -0.00 [m/s], Up = 0.03 [m/s]
New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 01 (Block IIF) with CN0=43.75 dB-Hz
New GPS NAV message received in channel 7: Satellite GPS PRN 01 (Block IIF) reports an unhealthy status, not used for navigation
subframe 3 from satellite GPS PRN 26 (Block IIF) with CN0=45.20 dB-Hz
New GPS NAV message received in channel 3: subframe 3 from satellite GPS PRN 31 (Block IIR-M) with CN0=43.56 dB-Hz
New GPS NAV message received in channel 2: subframe 3 from satellite GPS PRN 28 (Block IIR) with CN0=43.02 dB-Hz
New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 03 (Block IIF) with CN0=41.84 dB-Hz
Current receiver time: 1 min 18 s
Position at 2024-Jul-04 03:17:30.000000 UTC using 4 observations is Lat = 39.154836 [deg], Long = -77.190693 [deg], Height = 100.06 [m]
Velocity: East: 0.04 [m/s], North: -0.00 [m/s], Up = 0.00 [m/s]
Position at 2024-Jul-04 03:17:30.500000 UTC using 4 observations is Lat = 39.154852 [deg], Long = -77.190698 [deg], Height = 94.48 [m]
Velocity: East: 0.03 [m/s], North: -0.03 [m/s], Up = -0.12 [m/s]

jwmelto, 2024-07-05:
USB 3 is fine; I only meant faster than USB 1.

Your results look fine. Height (altitude) is not terribly accurate. First understand that the position is calculated in Earth-Centered, Earth-Fixed (ECEF) XYZ coordinates. The conversion from the Z axis to height is subject to the accuracy of the earth model which will not be precise at any given coordinate.

Note that your latitude measurement is varying by about 0.0001 degrees, or 11 meters. Your height is varying by about 14 meters. These are consistent.

(BTW, you might want to scrub your GPS coordinates from this message, for privacy reasons)

XueWei08, 2024-07-08:
> USB 3 is fine; I only meant faster than USB 1.
>
> Your results look fine. Height (altitude) is not terribly accurate. First understand that the position is calculated in Earth-Centered, Earth-Fixed [...]

Thanks for the explanation and reminder, I modified the location.
The latitude and longitude accuracy is good, but I have a problem with the altitude. I tested in another location, the latitude and longitude are consistent, and the change is in three or four decimal places. But the height can vary from 80 to 160, some may be over 3000, 4000, and some are even negative, and the antenna position does not change. The whole file is too long, I pasted some position results here:

{
  "type":  "Feature",
  "properties": {
       "name": "Locations generated by GNSS-SDR" 
   },
  "geometry": {
      "type": "MultiPoint",
      "coordinates": [
       [xx.26735596677088, xx.19207579713977, 111.18417298607528],
       [xx.26735952335257, xx.19208559037605, 111.76997008919716],
       [xx.26736564516011, xx.19209787159621, 107.80995272472501],
       [xx.26735153088389, xx.19211206699968, 106.10525671672076],
       [xx.26736237426036, xx.19211801961941, 108.03086932562292],
       [xx.26736867331131, xx.19212167268027, 109.88594551011920],
       [xx.26736067723701, xx.19211823858897, 108.14024646114558],
       [xx.26736472772728, xx.19210953395673, 112.69073569495231],
       [xx.26739059252348, xx.19207896324626, 137.08527510333806],
       [xx.26740706506443, xx.19207239920516, 141.19660758506507],
       [xx.26740672418943, xx.19209111492188, 142.18426396884024],
       [xx.26740901620458, xx.19208557613879, 141.52729683835059],
       [xx.26740138344809, xx.19207965435947, 143.14553072955459],
       [xx.26739359661430, xx.19208234250864, 143.20197472628206],
       [xx.26738522389455, xx.19209753399206, 139.52400929108262],
       [xx.26739628265034, xx.19208765345868, 139.46970343589783],
       [xx.26739002462217, xx.19209623900235, 136.05469939671457],
       [xx.26735835849668, xx.19205554055953, 126.01664137560874],
       [xx.26734181861968, xx.19206554659890, 123.69906718842685],
       [xx.26733729803452, xx.19204826150013, 120.66273934766650],
       [xx.26732849780066, xx.19207833026455, 120.72838379349560],
       [xx.26732760604582, xx.19209376526047, 124.60041280463338],
       [xx.26732859378848, xx.19210928910648, 124.53964068274945],
       [xx.26734767611528, xx.19213333869407, 126.90155224502087],
       [xx.26733923541472, xx.19212955997100, 126.37340864166617],
       [xx.26734035265913, xx.19213440210288, 127.42168565932661],
       [xx.26735422584740, xx.19214980609336, 129.56383449863642],
       [xx.26735843843007, xx.19215664379807, 127.10793136060238],
       [xx.26737231523234, xx.19219101277999, 130.68311595264822],
       [xx.26738213775702, xx.19216768841613, 129.88883757777512],
       [xx.26738404022498, xx.19217833695507, 128.86191855743527],
       [xx.26738640626915, xx.19218337916021, 130.33915980905294],
       [xx.26740375912239, xx.19217457779465, 136.83385547064245],
       [xx.26740239093598, xx.19216882226377, 139.19388369098306],
       [xx.26739304771493, xx.19217773895895, 141.13143596891314],
       [xx.26739115185855, xx.19216876647397, 143.09929428342730],
       [xx.26738447627902, xx.19218806322610, 144.71980182360858],
       [xx.26738754855431, xx.19218845030154, 149.29546264093369],
       [xx.26738256626703, xx.19215776977232, 149.03412929736078],
       [xx.26737465991006, xx.19215159032397, 151.97425260953605],
       [xx.26736575641526, xx.19215053539098, 150.66360297612846],
       [xx.26736814014663, xx.19215436173045, 155.55039367638528],
       [xx.26736423875278, xx.19215687298844, 160.04773772228509],
       [xx.26735410029738, xx.19214257801973, 152.17181441746652],
       [xx.26742057384568, xx.19207664188498, 191.95574209559709],
       [xx.26719779494380, xx.19242605983207, 186.27611701935530],
       [xx.26739733747829, xx.19278334541846, 290.61016690731049],
       [xx.26742937999401, xx.19294362660637, 291.64615640509874],
       [xx.26731243552658, xx.19297053113011, 251.07124020066112],
       [xx.26746317020154, xx.19349082204307, 353.87696195300668],
       [xx.26724445602014, xx.19390611688995, 388.87845224048942],
       [xx.26720682770092, xx.19432413516373, 417.97972937859595],
       [xx.26745158361150, xx.19498250697425, 563.29928435105830],
       [xx.26752878029114, xx.19527503273296, 653.27757317945361],
       [xx.26750249479609, xx.19543842627833, 716.78463949542493],
[... 93 lines cut ...]
       [xx.26746793986403, xx.19211787088749, 155.80894846282899],
       [xx.26747004243231, xx.19214347746788, 160.43867039307952],
       [xx.26747721596000, xx.19212387698633, 160.77388981636614],
       [xx.26746839411803, xx.19211153163208, 158.21656275354326],
       [xx.26744863711406, xx.19211969663120, 149.70602401252836],
       [xx.26744148725142, xx.19212355572274, 149.17985224444419],
       [xx.26744186348630, xx.19213409394461, 152.28477445151657],
       [xx.26741233321177, xx.19209863344607, 144.06780198402703],
       [xx.26740144559616, xx.19208375645687, 141.85483750142157],
       [xx.26739148126376, xx.19212122868843, 142.28866613376886],
       [xx.26741936389600, xx.19212342370542, 144.29608047660440],
       [xx.26738992605166, xx.19211154826643, 135.83905258961022],
       [xx.26739039992238, xx.19209743140537, 137.18754207715392],
       [xx.26735399002428, xx.19209934534857, 117.04063072428107],
[... cut ...]
       [xx.26728779069643, xx.19203783721553, 66.13246957771480],
       [xx.26729534570046, xx.19204292458905, 66.01768824178725],
       [xx.26729002936125, xx.19201402990848, 66.58622504491359],
       [xx.26730200894951, xx.19204418287819, 69.39463255275041],
       [xx.26731281882203, xx.19204293939762, 81.23538367450237],

Is this related to I change the place or the temperature outside is a little high(about 30℃ ) and affects the hardware performance? I am not sure. 
Do you have idea about the reason of it? or how can I avoid  or fix it? Thanks!

XiaoMiku01, 2024-08-13:
Hello! I am using the USRP B200 device just like you, and I have encountered the same issue. I heard that you have already resolved it and received the location information. Could you please share your configuration file with me? Thank you!

carlesfernandez, 2024-09-20:
Completed

closed 2024-09-20 by carlesfernandez