gnss-sdr/gnss-sdr #784: link errors with Position Independend Executable (PIE) mode

rsmegal, 2024-07-04:
gcc13-PIE - A default configuration to build all binaries in PIE mode is installed on opensuse Leap 15.6 and when building gnss-sdr (commit bb2d384ad6c2dfdb09f465a49e3925bd61fd9e6e) the following error reports are generated:

[ 95%] Linking CXX executable gnss-sdr
/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: ../core/libs/supl/libcore_libs_supl.a(supl.c.o): relocation R_X86_64_32 against hidden symbol `asn_DEF_ULP_PDU' can not be used when making a PIE object
/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status

/usr/lib64/gcc/x86_64-suse-linux/13/../../../../x86_64-suse-linux/bin/ld: ../algorithms/telemetry_decoder/libs/libswiftcnav/libtelemetry_decoder_libswiftcnav.a(cnav_msg.c.o): relocation R_X86_64_32 against `.bss' can not be used when making a PIE object; recompile with -fPIE

Any suggestions on the best way forward? I could deinstall gcc13-PIE without collateral damage to my system or I could switch to an alternative compiler; the system compiler is gcc 7.5.0.  Is there a fix to gnss-sdr build configuration?

jwmelto, 2024-07-04:
Just a guess here, but I tend to agree with [this SO answer](https://stackoverflow.com/a/19365454/9220132) ... "do what the compiler tells you" and compile with `-fPIE`.

It's also possible that CMake doesn't understand PIE yet. You could ask at https://discourse.cmake.org.

I think this is likely a case of when you are on the bleeding edge, you can cut yourself, but it's pure speculation.

rsmegal, 2024-07-06:
I am somewhat short of band aids at this time so I'll back away from the bleeding edge. I reverted to gcc (SUSE Linux) version 9.3.1 which is what I have been using for gnss-sdr up until now. The build works.

closed 2024-07-06 by rsmegal