gnss-sdr/gnss-sdr #785: Building GNSS-SDR from source error using m3 pro MacBook Pro

xin0606, 2024-07-05:
I'm currently try to build GNSS-SDR from source using M3 pro Macbook pro in Sonoma 14. I using homebrew to install all dependency with anaconda python environment.
However, the error occurs when I try to link gnss-sdr executable, the error messages are as below:

[image] 

[image]

Am I compiling in a wrong way?

jwmelto, 2024-07-05:
your first error is on `gr::sync_block` which suggests that the GNU Radio you installed is not M3 native. You might try building it from source as well, or check with the GNU Radio project.

carlesfernandez, 2024-08-09:
I can't reproduce this error in your very same hardware. I could build the software in the same architecture and OS (M3, Sonoma 14) without issues. Please review your GNU Radio installation.

closed 2024-08-09 by carlesfernandez