gnss-sdr/gnss-sdr #792: Unable to acquire L2C GPS with HackRF and gnss-sdr

tstellanova, 2024-08-10:
I've been able to create configuration files that apparently work with L1 C/A and L5 GPS, but so far I have not been able to create a configuration file that successfully acquires L2C.  I've tried modifying the  `/conf/gnss-sdr_GPS_L2C_USRP1_realtime.conf` for HackRF , and also starting with the L1 and L5 conf files and modifying for L2C.   

Typically gnss-sdr reports:
```
Current receiver time: 18 s
Tracking of GPS L2C signal started on channel 4 for satellite GPS PRN 32 (Block IIF)
...
Current receiver time: 46 s
OOOOOOOOOOOOOOOOOOOOCurrent receiver time: 47 s
OOOOOOOOOOOOOOOOOOOOOCurrent receiver time: 48 s
```
which I take to mean an overflow.   It's unclear to me why the L2C pipeline should overflow when the sampling rate is reduced from eg the L5 pipeline. 
I've attached several `.conf` files I've tried without success:
[hrf_gps_L2.conf.txt](https://github.com/user-attachments/files/16568278/hrf_gps_L2.conf.txt)
[hrf_gps_L2_trial.conf.txt](https://github.com/user-attachments/files/16568279/hrf_gps_L2_trial.conf.txt)
[hrf_gps_L2_orig.conf.txt](https://github.com/user-attachments/files/16568280/hrf_gps_L2_orig.conf.txt)

Thanks for any help. FWIW I've verified that the same antenna and bias-tee power supply works with a hardware-based L2 GPS receiver.

wkazubski, 2024-08-25:
L2C decoder has a lot of critical timing. I had to replace 4-core i5 CPU with 8-core i7 one and limit the number of channels to 5 to decode L2C in realtime. All available channels have to be locked to a signal to avoid overflows. You can also record the signal with GNU Radio and try to decode it offline.

carlesfernandez, 2024-08-30:
Hi,

could you try this configuration for the Acquisition block?

```
Acquisition_2S.implementation=GPS_L2_M_PCPS_Acquisition
Acquisition_2S.item_type=gr_complex
Acquisition_2S.doppler_max=4500
Acquisition_2S.doppler_step=125
Acquisition_2S.use_CFAR_algorithm=false
Acquisition_2S.threshold=10
Acquisition_2S.blocking=true
```

This conf is used in our CI to theck the L2C functionality. Hope this helps!

tstellanova, 2024-09-04:
@carlesfernandez Thanks for the example. I tried that `Acquisition_2S` conf block with the sample ` gps_l2c_m_prn7_5msps.dat` file but got no log output (as far as I could see) indicating successful acquisition.  Possibly the remainder of my conf file is misconfigured? Could someone provide a conf file that they expect should work with a sample dat file? 

Looking at the `GpsL2MPcpsAcquisitionTest` in  `/src/tests/unit-tests/signal-processing-blocks/acquisition/gps_l2_m_pcps_acquisition_test.cc` file on git `next` branch, it looks like it is using different Acquisition_2S configuration than those specified above.  I tried to reproduce this with a separate conf file, but with no luck.

Is there a sample dat file used in CI along with a sample conf file that allows reliable acquisition?

wkazubski, 2024-09-04:
Try with less channels configured, even with only one. If you get no overflows, signal lock and messages coming even from one satellite, your decoder settings are correct. Then increase the number of channels and try again.
I usually use only 5 channels because my antenna is on the windowsil and I can see only half of the sky,

tstellanova, 2024-09-05:
@wkazubski Thanks for the suggestion. I've reduced the number of channels to one with no luck, using a pre-recorded file source. 

I tried building for extended unit tests and running `GpsL2MPcpsAcquisitionTest`, which passes. So the next thing I'll try is to review carefully the configuration used by that test, try to create a matching conf file, then rerun gnss-sdr against a signal file source.

wkazubski, 2024-09-09:
This is the config file I used to receive L2C signal in realtime using RTL-SDR frontend. Dhe dongle used has a TCXO with some 2.5ppm offset so the frequency correction is not needed.  
[gnss-sdr_GPS_L2C_rtlsdr_realtime.conf.txt](https://github.com/user-attachments/files/16929804/gnss-sdr_GPS_L2C_rtlsdr_realtime.conf.txt)

tstellanova, 2024-09-10:
Thanks for the sample config file. I swapped in an RTLSDR and was able to obtain a position fix after about ten minutes. I'm going to close this issue and assume that I can eventually tweak the HackRF conf to obtain a fix.

closed 2024-09-10 by tstellanova