gnss-sdr/gnss-sdr #794: *** stack smashing detected ***: terminated

cl-cv, 2024-08-15:
I just did a clean build of gnss-sdr on a new VM. I had no problems in the build process, I ran the VOLK profilers, etc, but when trying to run a test (which I have working on my other VM), I get:

```
Initializing GNSS-SDR v0.0.19.git-- ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
*** stack smashing detected ***: terminated
Aborted
```

I only found [one issue](https://github.com/gnss-sdr/gnss-sdr/issues/653) referencing this, and it doesn't actually address my problem (as far as I know). Why is this occurring?

carlesfernandez, 2024-08-30:
Hi @ci-cv, 

could you provide details about what test are you trying to run, and what was the commit you are working with, and what OS distribution and version are you using? 

Thank you.

carlesfernandez, 2024-10-03:
Please feel free to open a new issue if the problem remains.

closed 2024-10-03 by carlesfernandez