gnss-sdr/gnss-sdr #799: Segmentation fault (core dumped)

aurora211e, 2024-08-26:
Hi,
I'm trying to get through the "My first position fix" tutorial, but I'm facing the following error:
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.9 ... Please wait.
Logging will be done at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/aurora/work/data/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
CONFIGURATION WARNING: Parameters GNSS-SDR.internal_fs_hz and Resampler.sample_freq_out are not set to the same value!
E0826 04:15:42.479465  1852 gnss_block_factory.cc:1112] PVT.RTKLIB_PVT: Undefined implementation for block

carlesfernandez, 2024-08-26:
Please note that you are running a very old version of GNSS-SDR. Please uninstall the package and build it from sources:

```
git clone https://github.com/gnss-sdr/gnss-sdr/
cd gnss-sdr/build
git checkout next
cmake .. && make && sudo make install
```

aurora211e, 2024-08-26:
> 请注意，您运行的是非常旧版本的 GNSS-SDR。请卸载该包并从源代码构建它：
> 
> ```
> git clone https://github.com/gnss-sdr/gnss-sdr/
> cd gnss-sdr/build
> git checkout next
> cmake .. && make && sudo make install
> ```

fine,i'll try it

aurora211e, 2024-08-27:
> > 请注意，您运行的是非常旧版本的 GNSS-SDR。请卸载该包并从源代码构建它：
> > ```
> > git clone https://github.com/gnss-sdr/gnss-sdr/
> > cd gnss-sdr/build
> > git checkout next
> > cmake .. && make && sudo make install
> > ```
> 
> fine,i'll try it

git clone https://github.com/gnss-sdr/gnss-sdr/
Cloning into 'gnss-sdr'...
fatal: unable to access 'https://github.com/gnss-sdr/gnss-sdr/': Failed to connect to github.com port 443: Connection refused
this is a new question,how can i resolve it?

aurora211e, 2024-08-27:
> Please note that you are running a very old version of GNSS-SDR. Please uninstall the package and build it from sources:
> 
> ```
> git clone https://github.com/gnss-sdr/gnss-sdr/
> cd gnss-sdr/build
> git checkout next
> cmake .. && make && sudo make install
> ```

c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
src/algorithms/PVT/gnuradio_blocks/CMakeFiles/pvt_gr_blocks.dir/build.make:62: recipe for target 'src/algorithms/PVT/gnuradio_blocks/CMakeFiles/pvt_gr_blocks.dir/rtklib_pvt_gs.cc.o' failed
make[2]: *** [src/algorithms/PVT/gnuradio_blocks/CMakeFiles/pvt_gr_blocks.dir/rtklib_pvt_gs.cc.o] Error 4
CMakeFiles/Makefile2:2327: recipe for target 'src/algorithms/PVT/gnuradio_blocks/CMakeFiles/pvt_gr_blocks.dir/all' failed
make[1]: *** [src/algorithms/PVT/gnuradio_blocks/CMakeFiles/pvt_gr_blocks.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
i have a new question,how slove it

carlesfernandez, 2024-09-02:
Hi, 

```
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-7/README.Bugs for instructions.
```
This error means that your computer ran out of memory when compiling. Do no use `make -j ...`.

Then:
```
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
```

This error can happen in a slow internet connection whren cloning a git repository. Incrementing the buffer size could fix this issue:

```
git config --global http.postBuffer 524288000
```

Hope this helps.

closed 2024-09-13 by carlesfernandez