gnss-sdr/gnss-sdr #800: Unable to Acquire Galileo E6B/HAS PVT via Next Branch - Config File Issue?

LRParser, 2024-09-04:
First - thanks to the authors and maintainers for a really excellent piece of software! This tool works great with a HackRF One and active multi-band GPS antenna to receive GPS L1 position fixes.

I wanted to take the next step up in complexity and try to receive E6B. I used a standalone development board (ArduSimple RTK3B Budget) and confirmed it and the antenna could receive E6B/Galileo HAS well. Next I tried to receive E6B/HAS using HackRF One, the same antenna, and gnss-sdr `next` branch.

I can get as far as seeing "Tracking..." etc in the logs, but cannot get further. Is there something else I should try? Is it normal for a fix to take a long time, and I should wait longer? I disconnected after around 3 minutes, as L1 fix was quite fast for me.

```
joe@joe-msivr:~/gnss-sdr/test$ gnss-sdr -c e6.conf 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.19.git-next-1088387f7 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2024.02.1 
Enabled antenna bias voltage
Actual RX Rate: 4e+06 [SPS]...
Actual RX Freq: 1.27875e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 14 dB...
Actual Bandwidth: 6e+06 [Hz]...
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
TcpCmdInterface: Telecommand TCP interface listening on port 3333
Current receiver time: 1 s
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
Current receiver time: 14 s
Current receiver time: 15 s
Current receiver time: 16 s
Current receiver time: 17 s
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
Current receiver time: 21 s
Current receiver time: 22 s
Current receiver time: 23 s
Current receiver time: 24 s
Current receiver time: 25 s
Current receiver time: 26 s
Current receiver time: 27 s
Current receiver time: 28 s
Current receiver time: 29 s
Current receiver time: 30 s
Current receiver time: 31 s
Current receiver time: 32 s
Current receiver time: 33 s
Current receiver time: 34 s
Current receiver time: 35 s
Current receiver time: 36 s
Current receiver time: 37 s
Current receiver time: 38 s
Current receiver time: 39 s
Current receiver time: 40 s
Current receiver time: 41 s
Current receiver time: 42 s
Current receiver time: 43 s
[... 204 lines cut ...]
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1111

Monitor.enable_monitor=true
Monitor.decimation_factor=1
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1112

GNSS-SDR.telecommand_enabled=true
GNSS-SDR.telecommand_tcp_port=3333

``` 
Is there anything I can further log to assist in debugging? I could not find a full E6B config file anywhere in the documentation or in the source, but will be happy to PR one if I can get it working.

carlesfernandez, 2024-09-09:
Dear @LRParser ,

thanks for your detailed report. Since the Galileo E6B signal has a spreading code chipping rate of 5.115 MChip/s, you will need to increase your bandwidth (that is, `GNSS-SDR.internal_fs_sps` and `SignalSource.sampling_frequency`), so you get at least one complex sample per chip.

The rest of the configuration looks good to me.

Hope this helps,
Carles

LRParser, 2024-09-09:
Thanks Carles! By setting `GNSS-SDR.internal_fs_sps` and `SignalSource.sampling_frequency` to `6000000` I can now receive E6B HAS pages, but after 25 minutes I am still not getting a PVT lock. I think this must be because E6B is only sending corrections but cannot be used alone for positioning, is my understanding right here? I can see a `HAS_Messages` document that is filling up, but I still do not get a position fix.

If I add a second receiver/HackRF and configure a SignalSource0 at E6B and a SignalSource1 at an L1 frequency, will the `gnss-sdr` PVT code know how to apply the corrections to the L1 signal using the data from E6B? Or is this something I need to do outside of the `gnss-sdr` codebase, e.g. with `RTKLib`or `HASLib`?

carlesfernandez, 2024-09-09:
Hi @LRParser,

the E6B signal does not convey a TOW, so you can't compute pseudoranges with this signal alone. You need to combine it with another Galileo signal (e.g., E1) in order to obtain PVT fixes. In those multi-band configurations, gnss-sdr will apply the HAS corrections to the delivered PVT fixes.

closed 2024-09-13 by carlesfernandez