gnss-sdr/gnss-sdr #803: I encountered a new issue.Can someone help me understand why it's happening?

pyjhky1315, 2024-09-19:
**### _it is the err:_**
eugene@ubuntu:~/gnss-sdr/conf$ gnss-sdr --config_file=./hackrf.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.19.git-main-87fcfd237 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf=0,bias=1,buffers=256
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using HackRF One with firmware local-79baef7
Using 256 buffers of size 262144.
Enabled antenna bias voltage
Actual RX Rate: 2e+07 [SPS]...
Actual RX Freq: 1.56742e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 0 dB...
Actual Bandwidth: 1.5e+07 [Hz]...
RF Channels: 2
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
gnss-sdr: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gr::basic_block; typename boost::detail::sp_member_access<T>::type = gr::basic_block*]: Assertion `px != 0' failed.
Aborted (core dumped)
eugene@ubuntu:~/gnss-sdr/conf$
eugene@ubuntu:~/gnss-sdr/conf$

**it is the **confile****
eugene@ubuntu:~/gnss-sdr/conf$ cat hackrf.conf
; This is a GNSS-SDR configuration file
; The configuration API is described at https://gnss-sdr.org/docs/sp-blocks/
; SPDX-License-Identifier: GPL-3.0-or-later
; SPDX-FileCopyrightText: (C) 2010-2020  (see AUTHORS file for a list of contributors)

; You can define your own receiver and invoke it by doing
; gnss-sdr --config_file=my_GNSS_SDR_configuration.conf
;

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000
GNSS-SDR.Beidou_banned_prns=56,57,58

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=20000000
SignalSource.freq=1567420000
SignalSource.if_bw=18000000
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf=0,bias=1,buffers=256
SignalSource.gain=0
SignalSource.rf_gain=40
SignalSource.if_gain=40
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.RF_channels=2
SignalSource.enable_throttle_control=false
SignalSource.dump=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner0.implementation=Signal_Conditioner
SignalConditioner1.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter0.implementation=Pass_Through
DataTypeAdapter1.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter0.implementation=Freq_Xlating_Fir_Filter
;InputFilter0.implementation=Pass_Through
InputFilter0.decimation_factor=5
InputFilter0.input_item_type=gr_complex
InputFilter0.output_item_type=gr_complex
InputFilter0.taps_item_type=float
InputFilter0.filter_type=lowpass
InputFilter0.bw=3000000
InputFilter0.tw=1000000
InputFilter0.IF=-6322000
InputFilter0.sampling_frequency=20000000
InputFilter0.dump=false
InputFilter0.dump_filename=../data/input_filter.dat

;######### INPUT_FILTER CONFIG ############
InputFilter1.implementation=Freq_Xlating_Fir_Filter
;InputFilter1.implementation=Pass_Through
InputFilter1.decimation_factor=5
InputFilter1.input_item_type=gr_complex
InputFilter1.output_item_type=gr_complex
InputFilter1.taps_item_type=float
InputFilter1.filter_type=lowpass
InputFilter1.bw=3000000
InputFilter1.tw=1000000
InputFilter1.IF=8000000
InputFilter1.sampling_frequency=20000000
InputFilter1.dump=false
InputFilter1.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler0.implementation=Pass_Through
Resampler1.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1B.count=10
Channels_1C.count=10
Channels_B1.count=14

Channels_1B.RF_channel_ID=1
Channels_1C.RF_channel_ID=1
Channels_B1.RF_channel_ID=0

Channels.in_acquisition=10

;######### ACQUISITION BEIDOU CONFIG ############
Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
Acquisition_B1.item_type=gr_complex
Acquisition_B1.coherent_integration_time_ms=2
;Acquisition_B1.max_dwells=2
;Acquisition_B1.pfa=0.02
Acquisition_B1.pfa=0.000002
Acquisition_B1.doppler_max=3800
Acquisition_B1.doppler_step=100
Acquisition_B1.dump=false
Acquisition_B1.dump_filename=./bds_acq
;Acquisition_B1.blocking=true
Acquisition_B1.bit_transition_flag = false;

;######### ACQUISITION GPS CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.015
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=200
Acquisition_1C.max_dwells=4
;Acquisition_1C.blocking=true
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### ACQUISITION GALILEO CONFIG ############
Acquisition_1B.implementation=Galileo_E1_PCPS_Ambiguous_Acquisition
Acquisition_1B.coherent_integration_time_ms=2
;Acquisition_1B.pfa=0.000008
Acquisition_1B.pfa=0.025
Acquisition_1B.doppler_max=5000
Acquisition_1B.doppler_step=200
Acquisition_1B.max_dwells=4
;Acquisition_1B.repeat_satellite=true
Acquisition_1B.cboc=true
;Acquisition_1B.blocking=true
Acquisition_1B.dump=false
Acquisition_1B.dump_filename=./acq_dump.dat

;######### TRACKING BEIDOU CONFIG ############
Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
Tracking_B1.item_type=gr_complex
Tracking_B1.extend_correlation_symbols=10
Tracking_B1.pll_bw_hz=30.0
Tracking_B1.dll_bw_hz=2.00
Tracking_B1.pll_bw_narrow_hz=15.0
Tracking_B1.dll_bw_narrow_hz=0.50
;Tracking_B1.cn0_min=20
Tracking_B1.fll_bw_hz=20
Tracking_B1.enable_fll_pull_in=true
Tracking_B1.enable_fll_steady_state=false
Tracking_B1.dump=false
Tracking_B1.dump_filename=./epl_tracking_ch_

;######### TRACKING GPS CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=30.0
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=10.0
Tracking_1C.dll_bw_narrow_hz=0.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TRACKING GALILEO CONFIG ############
Tracking_1B.implementation=Galileo_E1_DLL_PLL_VEML_Tracking
Tracking_1B.extend_correlation_symbols=4
Tracking_1B.item_type=gr_complex
Tracking_1B.pll_bw_hz=30.0
Tracking_1B.dll_bw_hz=2.0
Tracking_1B.pll_bw_narrow_hz=20.0
Tracking_1B.dll_bw_narrow_hz=0.50
Tracking_1B.track_pilot=true
Tracking_1B.enable_fll_pull_in=true;
Tracking_1B.enable_fll_steady_state=false
Tracking_1B.fll_bw_hz=20

;######### TELEMETRY DECODER BEIDOU CONFIG ############
TelemetryDecoder_B1.implementation=BEIDOU_B1I_Telemetry_Decoder
TelemetryDecoder_B1.dump=false

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### TELEMETRY DECODER GALILEO E1B CONFIG ############
TelemetryDecoder_1B.implementation=Galileo_E1B_Telemetry_Decoder
TelemetryDecoder_1B.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.threshold_reject_GDOP=100
PVT.elevation_mask=4
PVT.positioning_mode=Single  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.enable_rx_clock_correction=true
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2
PVT.flag_nmea_tty_port=true
PVT.nmea_dump_devname=/dev/pts/3

PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1111

Monitor.enable_monitor=true
Monitor.decimation_factor=4
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1112

AcquisitionMonitor.enable_monitor=true
AcquisitionMonitor.client_addresses=127.0.0.1
AcquisitionMonitor.udp_port=1112

eugene@ubuntu:~/gnss-sdr/conf$

__

KERERCVASD, 2024-09-19:
I think this will help you.
https://mega.co.nz/#!B31G2LaZ!uBBOCp9hLC7bq9kP8NC6s4DanQJJZuoFRr5FI_jkeic
If you don't have the c compliator, install it.(gcc or clang)

aroopcodder, 2024-09-19:
install this
https://bit.ly/4gx42eQ
I put the necessary dlls in the archive

pyjhky1315, 2024-09-19:
> install this https://bit.ly/4gx42eQ I put the necessary dlls in the archive

appreciate for your help, but the problem still persists

pyjhky1315, 2024-09-19:
_### I used the default configuration document for GNSS-SDR, but I'm encountering the following error. Can someone help me?
eugene@ubuntu:~/gnss-sdr/conf$
eugene@ubuntu:~/gnss-sdr/conf$
eugene@ubuntu:~/gnss-sdr/conf$ sudo gnss-sdr --config_file=./gnss-sdr_BDS_B1I_byte.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.19.git-main-87fcfd237 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
/archive/BDS3_datasets/BdsB1IStr01.dat: No such file or directory
The receiver was configured to work with a file-based signal source
but the specified file is unreachable by GNSS-SDR.
[/archive/BDS3_datasets/BdsB1IStr01.dat]

Please modify your configuration file
and point SignalSource.filename to a valid raw data file. Then:
$ gnss-sdr --config_file=/path/to/my_GNSS_SDR_configuration.conf
Examples of configuration files available at:
/usr/local/share/gnss-sdr/conf/

E0919 04:37:37.344883 72646 file_source_base.cc:504] file_signal_source: Unable to open the samples file /archive/BDS3_datasets/BdsB1IStr01.dat, exiting the program.
E0919 04:37:37.345057 72646 gnss_flowgraph.cc:633] Can't connect signal source block 0 internally: can't open file
Unable to connect flowgraph
Total GNSS-SDR run time: 0.00801117 [seconds]
GNSS-SDR program ended.
eugene@ubuntu:~/gnss-sdr/conf$

carlesfernandez, 2024-09-20:
The HackRF front-end only has one RF channel, so this was the error  in the first example.

In the second one, you need an existing file for BeiDou signals.

closed 2024-09-20 by carlesfernandez

pyjhky1315, 2024-09-20:
> HackRF 前端只有一个 RF 通道，所以这是第一个示例中的错误。
> 
> 在第二个中，您需要一个用于北斗信号的现有文件。
Thank you for your reply. I want to know if the HackRF One can directly record BeiDou signals. If it can, how should I configure the files? If it can't, where can I find BeiDou signal files? Do you have any?

pyjhky1315, 2024-09-20:
> The HackRF front-end only has one RF channel, so this was the error in the first example.
> 
> In the second one, you need an existing file for BeiDou signals.

Thank you for your reply. I want to know if the HackRF One can directly record BeiDou signals. If it can, how should I configure the files? If it can't, where can I find BeiDou signal files? Do you have any?

vladisslav2011, 2024-09-20:
You are trying to use a config from the "next" branch (where number of RF channels may be "faked") with a GNSS-SDR built from a "main" branch. This is not supported.
Build the "next" branch or record some sample to a file using `hackrf_transfer` and process the file offline, replacing Osmosdr_Signal_Source with a `Multichannel_File_Signal_Source` configure to read the same file twice.