gnss-sdr/gnss-sdr #807: Mutiple band in usrp B210

Doncher7, 2024-09-26:
I want to use the two RX front ends of the USRP B210 to collect the signals of the Beidou B1I and GPS L1 in real time. However, when it is running, it is found that the device can only receive GPS NAV message, and Beidou's satellite signal is always in the capture stage. However, when I run the config of Beidou or GPS alone, I can receive the corresponding signal and complete the positioning.
[image]

Doncher7, 2024-09-26:
this is my config content

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2500000  ; 根据 USRP 的实际能力设置采样率
;GNSS-SDR.num_sources=2

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.RF_channels=2

SignalSource.implementation=UHD_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2500000  ; 设为2 MSps
SignalSource.subdevice=A:A A:B  ; USRP B210 的默认子设备

SignalSource.freq0=1561098000  ; 北斗 B1I 信号的中心频率为 1561.098 MHz
SignalSource.freq1=1575420000
SignalSource.IF_bandwidth_hz0 = 1000000

SignalSource.gain0=60  ; 根据需要调整增益
SignalSource.gain1=40  ; 根据需要调整增益

SignalSource.samples0=0  ; 实时处理
SignalSource.samples1=0  ; 实时处理

SignalSource.dump0=false
SignalSource.dump1=false

SignalSource.dump_filename0=./signal_source0.dat
SignalSource.dump_filename1=./signal_source1.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner0.implementation=Signal_Conditioner
SignalConditioner1.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=gr_complex

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Fir_Filter
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=11
InputFilter.number_of_bands=2

InputFilter.band1_begin=0.0
InputFilter.band1_end=0.48
InputFilter.band2_begin=0.52
InputFilter.band2_end=1.0

InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0

InputFilter.band1_error=1.0
InputFilter.band2_error=1.0

InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=2500000
InputFilter.IF=0

;######### RESAMPLER CONFIG ############
Resampler0.implementation=Pass_Through
Resampler1.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_B1.count=8
Channels_1C.count=8
Channels.in_acquisition=2

Channel0.RF_channel_ID=0
Channel1.RF_channel_ID=0
Channel2.RF_channel_ID=0
Channel3.RF_channel_ID=0
Channel4.RF_channel_ID=0
Channel5.RF_channel_ID=0
Channel6.RF_channel_ID=0
Channel7.RF_channel_ID=0
Channel8.RF_channel_ID=1
Channel9.RF_channel_ID=1
Channel10.RF_channel_ID=1
Channel11.RF_channel_ID=1
Channel12.RF_channel_ID=1
Channel13.RF_channel_ID=1
Channel14.RF_channel_ID=1
Channel15.RF_channel_ID=1

Channel0.signal=B1
Channel1.signal=B1
Channel2.signal=B1
Channel3.signal=B1
Channel4.signal=B1
Channel5.signal=B1
Channel6.signal=B1
Channel7.signal=B1
Channel8.signal=1C
Channel9.signal=1C
Channel10.signal=1C
Channel11.signal=1C
Channel12.signal=1C
Channel13.signal=1C
Channel14.signal=1C
Channel15.signal=1C

;Channel0.satellite = 6;
;Channel1.satellite = 8;
;Channel2.satellite = 9;
;Channel3.satellite = 13;
;Channel4.satellite = 17;
;Channel5.satellite = 1;
;Channel6.satellite = 2;
;Channel7.satellite = 3;
;Channel8.satellite = 4;
;Channel9.satellite = 5;

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
Acquisition_B1.item_type=gr_complex
Acquisition_B1.coherent_integration_time_ms=1
Acquisition_B1.pfa=0.01
Acquisition_B1.doppler_max=10000
Acquisition_B1.doppler_step=500
Acquisition_B1.dump=false
Acquisition_B1.blocking=false
Acquisition_B1.bit_transition_flag=false

;######### TRACKING GLOBAL CONFIG ############
Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
Tracking_B1.item_type=gr_complex
Tracking_B1.pll_bw_hz=25.0
Tracking_B1.dll_bw_hz=2.5
Tracking_B1.dump=false

;######### TELEMETRY DECODER B1 CONFIG ############
TelemetryDecoder_B1.implementation=BEIDOU_B1I_Telemetry_Decoder
TelemetryDecoder_B1.dump=false

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=40
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=false
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.nmea_dump_filename=./gnss_sdr_bds_b1i_pvt.nmea
PVT.flag_nmea_tty_port=false
PVT.flag_rtcm_server=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=true
PVT.dump_filename=./PVT
PVT.rinex_output_enabled=true
PVT.gpx_output_enabled=true
PVT.rinex_version=2

closed 2024-10-03 by carlesfernandez

Doncher7, 2024-10-08:
> As far as I know, the USRP B210 does not have the capability of tuning two inputs at different frequencies.

Thank you. I'll try it with another device

Doncher7, 2024-10-08:
> As far as I know, the USRP B210 does not have the capability of tuning two inputs at different frequencies.

Can the N310 capture satellite signals of different frequencies?

carlesfernandez, 2024-10-08:
The N310 RF front end uses two [AD9371](https://www.analog.com/en/products/ad9371.html) transceivers, so yes, you can tune it to two GNSS frequency bands.