gnss-sdr/gnss-sdr #812: Rinex2assist (Segmentation fault (core dumped))

Derso1, 2024-10-07:
Hi!
I did the procedure to install and run rinex2assist, but when I run "rinex2assist chpg0940.13.z " I have this message: "Segmentation fault (core dumped)"
I did some searches in the internet, but no one have the same problem.
Can anyone help-me?

carlesfernandez, 2024-10-11:
Could you describe your environment, so we can reproduce the error?

Zi7ar21, 2024-10-12:
I don't know what procedure @Derso1 followed to install `rinex2assist`, but [it's only compiled if ENABLE_UNIT_TESTING_EXTRA, ENABLE_SYSTEM_TESTING_EXTRA, or ENABLE_FPGA was enabled at build time](https://github.com/gnss-sdr/gnss-sdr/blob/87fcfd237d5937ab2c193a1d68d37bf72653bcec/src/utils/CMakeLists.txt#L10-L13):

```cmake
if(ENABLE_UNIT_TESTING_EXTRA OR ENABLE_SYSTEM_TESTING_EXTRA OR ENABLE_FPGA)
    add_subdirectory(rinex-tools)
    add_subdirectory(rinex2assist)
endif()
```

There was [a conversation on the GNSS-SDR Matrix channel about problems building `rinex2assist`](https://matrix.to/#/!nHfhBhNcLjliIheyjH:gnuradio.org/$EeXYYKE-7ilKCxGpgb5FmKePacDuAqYEw3eIcZthPg0?via=gnuradio.org&via=matrix.org&via=arcticfoxes.net) which you may find useful. If you are having trouble getting assist data, you might also find looking at issue #721 helpful.

Derso1, 2024-10-14:
Hi everyone!

First of all, the procedure that I put above is this one (https://github.com/gnss-sdr/gnss-sdr/blob/main/src/utils/rinex2assist/README.md)

I tried it using my notebook and my desktop.

When I used my notebook the result was "Segmentation fault (core dumped)", but when I used my desktop everything worked fine.

In both of cases I am using Current release: [v0.0.19]

My Operational System is:

Notebook - Linux 6.8.0-45-generic #45~22.04.1 Ubuntu SMP PREEMPT_DYNAMICX86_64 x86_64_x86_64 GNU/Linux

DeskTop - Linux 6.8.0-45-generic #45~22.04.1 Ubuntu SMP PREEMPT_DYNAMICX86_64 x86_64_x86_64 GNU/Linux

If @carlesfernandez need more information I can prepare without problem.

TKS

carlesfernandez, 2024-10-25:
Maybe you can use `ldd` or `lddtree` to check what libraries are you linking against, and check the differences between the notebook and the desktop.