gnss-sdr/gnss-sdr #813: Always losing lock after being locked

ryanf2110, 2024-10-08:
I used a computer, usrp2922, and an active antenna to run GNSS SDR. I followed all the procedures on the official website, but after locking, there is always a situation where the lock is lost. I would like to know the reason for this.
Here are some relevant pictures and information, I need your help ,thank you very much!
[image]
[image]
[image]
[image]
[image]
[image]

Derso1, 2024-10-08:
Hi, 
I saw your test and a have some ideas:

1) Put bias-T near the front-end. I put my bias-t face-to-face.

2)Try to put PVT.enable_rx_clock_corretion=true

Derso1, 2024-10-08:
Additionally, I had many problems when I was using biass T, then we put a TEEL-1 4000; 
I had a good  improvement when we put this device

vladisslav2011, 2024-10-08:
Hello.
Your configuration is not correct.
I suggest following changes:
1. Set `SignalSource.sampling_frequency` to 4000000 to match `GNSS-SDR.internal_fs_sps` or set `GNSS-SDR.internal_fs_sps` to 2000000 to match `SignalSource.sampling_frequency`
2. Set `InputFilter.implementation` to "Pass_Through"
3. Set `InputFilter0.input_item_type` to "gr_complex"
4. Set `SignalSource.item_type` to "gr_complex"
5. Remove `DataTypeAdapter.item_type` line

ryanf2110, 2024-10-09:
> Hello. Your configuration is not correct. I suggest following changes:
>
> 1. Set `SignalSource.sampling_frequency` to 4000000 to match `GNSS-SDR.internal_fs_sps` or set `GNSS-SDR.internal_fs_sps` to [...]

Thank you very much for your answer. Here is my current configuration file, which can receive satellite signals with PRN=24. Except for this, all other satellites are unlocked，and below are the satellite signals I received using my mobile GPS test

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.device_address=192.168.10.2                                         
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.subdevice=A:0                                
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Pass_Through
InputFilter0.input_item_type=gr_complex
InputFilter.input_item_type=cshort
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=11
InputFilter.number_of_bands=2

InputFilter.band1_begin=0.0
InputFilter.band1_end=0.48
InputFilter.band2_begin=0.52
InputFilter.band2_end=1.0

InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0

InputFilter.band1_error=1.0
InputFilter.band2_error=1.0

InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=4000000
InputFilter.IF=0

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=40
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.enable_rx_clock_corretion=true
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

[image]
[image]

carlesfernandez, 2024-10-11:
We suggest you to change the sampling frequency to 4000000 Sps (please change both the SIgnalSOurce parameter and the global parameter at the top of the config file). Besides this, it seems that the satellite signal reaches the receiver with low power (38 dBHz is quite low for an open sky receiver).  

You can try to reduce the Tracking_1C.extend_correlation_symbol (e.g. 1 ms).

Another solution to increase the signal quality could be to insert an extra LNA between the antenna and the USRP.

ryanf2110, 2024-11-11:
> We suggest you to change the sampling frequency to 4000000 Sps (please change both the SIgnalSOurce parameter and the global parameter at the top of the config file). Besides this, it seems that the [...]

thanks for your reply! I have learned a lot,and I realized that weak signals and multipath signals seem to be difficult to receive. I want to study the reception of multipath signals,is it possible? if possible,how can I receive the multipath signals,waiting for your help