gnss-sdr/gnss-sdr #816: Stop the screen messages

Derso1, 2024-10-14:
Hi every one

i have a question:

there is some config to stop all the screen messages?

I am sending the data that I need using the serial port and then I would like to stop the screen msg.

tks

jwmelto, 2024-10-15:
There is not. You can redirect stdout to a file (or `/dev/null` but it cannot be disabled.

Derso1, 2024-10-15:
Tks a lot!!! @jwmelto

Derso1, 2024-10-15:
@jwmelto, A question:
When you told me: "it cannot be disabled", What is it mean?

1) It cannot because gnss-sdr does not have a config to do this, or

2) It cannot because if I disable the message I will creat a problem inside the code?

Tks

jwmelto, 2024-10-15:
It cannot be disabled because the code is littered with print statements to stdout. They are not conditional on some configuration parameter. 

As I said, you can redirect stdout somewhere, but the code will print always.

carlesfernandez, 2024-10-15:
Hi @Derso1,

There is no configuration option for this at the moment. The messages are currently generated using a simple `"std::cout << "I'm a message\n"` pattern in the source code, so redirecting output to `/dev/null` should work.

That said, I agree that this is a valid use case. Not all users need or want to see these messages. It might be worth considering a more consistent logging mechanism that gives users finer control. We could take advantage of the severity levels defined by the [Glog](https://google.github.io/glog/stable/logging/) and [Abseil Logging](https://abseil.io/docs/cpp/guides/logging) libraries to achieve this. While Glog has been our logging library since the start of the project, we're gradually transitioning to Abseil Logging, which seems like the natural successor. Glog will continue to be supported with no set deprecation timeline. Fortunately, both libraries share similar concepts, which could allows us to offer better control over application logging.

Any thoughts on this, @jwmelto ?

Derso1, 2024-10-15:
Thank every one!
In my case, I am trying to embedded gnss-sdr.
First of all I got the information that I needed and sent to a serial port. 
After this, my cowork and me rebuild the track using the information that we sent to serial port during a car test (it was very good).
Now, to embedded the code, We are withdrawing the screen msg.

jwmelto, 2024-10-18:
I would suggest moving to [spdlog](https://github.com/gabime/spdlog), since that's what GNU Radio uses. We might as well be consistent with the framework on which we are built.

I have also wanted the elimination of `stdout` but my investigation of the code suggests that `stdout` code fulfills a different role than logged output. It doesn't HAVE to, but I wasn't willing to make that change unilaterally.

mmajoral, 2024-12-20:
Thank you for your comments. We are not planning to provide an option to disable the screen messages in the short term. You have the option to redirect the messages to /dev/null or to a file descriptor.

closed 2024-12-20 by mmajoral