gnss-sdr/gnss-sdr #818: GNSS-SDR Embedded

Derso1, 2024-10-17:
Hello

I am trying to put the GNSS-SDR in one Toradex.  

It was possible to do this, but I am in the limit of my HW. 

In this way, I would like to know if there is any configuration that can help to increase my embedded performance.

What I did:

1) I the cmake file I put -DENABLE_PACKAGING==ON

2) I disabled all the screen msg

3) I configured observable_interval_ms=40

4) I configured  output_rate_ms=1000

Using this It was possible to get 7 channels, but I needed 8 channels....

Someone has any ideia that can help-me

TKS

Derso1, 2024-11-25:
Hi @carlesfernandez . I tries, but I did not see improvement. 
do you have some ideia?

TKS

mmajoral, 2024-12-20:
Hi @Derso1, what sampling frequency are you using? a possible solution to decrease the CPU load is to reduce the sampling frequency to the minimum value needed to process the GNSS signals.

Derso1, 2024-12-20:
Hi @mmajoral, I am using 1Hz. I think that the way to solve this will be to use a FPGA.