gnss-sdr/gnss-sdr #820: Can GNSS-SDR achieve four modes of satellite positioning

ryanf2110, 2024-10-21:
Can I use one USRP device to achieve satellite positioning in four modes (GPS, BD, Galileo, GLONASS)

vladisslav2011, 2024-10-21:
Yes. You can.

ryanf2110, 2024-10-22:
> Yes. You can.

Is it possible for me to use only one RF front-end for the USRP? I'm not sure

ryanf2110, 2024-10-22:
Can someone provide four modes of satellite positioning configuration files? thanks a lot

vladisslav2011, 2024-10-22:
Config:
https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/RealTime_input/gnss-sdr_multichannel_GPS_Galileo_Beidou_Glonass_L1_USRP_realtime.conf
Demo:
https://www.youtube.com/watch?v=HnZkKj9a-QM
Don't forget to build the `next` branch.

ryanf2110, 2024-10-23:
> Config: https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/RealTime_input/gnss-sdr_multichannel_GPS_Galileo_Beidou_Glonass_L1_USRP_realtime.conf Demo: https://www.youtube.com/watch?v=HnZkKj9a-QM Don't forget to build the `next` branch.

ryan@ubuntu:~/work/data$ gnss-sdr --config_file=/home/ryan/work/data/my_GNSS_receiver.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.19.git-next-d40e3427d ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes
Configuration error in SignalSource block, implementation UHD_Signal_Source. The error was:
LookupError: IndexError: multi_usrp: RX channel 1 out of range for configured RX frontends
GNSS-SDR program ended.

The above is the running error message. I am using USRP2922

vladisslav2011, 2024-10-23:
It looks like USRP2922 does not have enough bandwidth.
The bandwidth, required to receive all 4 systems at L1 is (1602+6)-(1561.098-3)=~50MHz, but USR2922 maximum bandwidth is 40MHz. Try using B200/BladeRF or something similar.
The error message is just saying that UHD_Signal_Source does not support overriding channel number with `SignalSource.RF_channels` config option. Try switching to Osmosdr_Signal_Source.

ryanf2110, 2024-11-11:
> It looks like USRP2922 does not have enough bandwidth. The bandwidth, required to receive all 4 systems at L1 is (1602+6)-(1561.098-3)=~50MHz, but USR2922 maximum bandwidth is 40MHz. Try using [...]

I have learned a lot,and  I realized that weak signals and multipath signals seem to be difficult to receive. I want to study the reception of multipath signals,is it possible?

carlesfernandez, 2024-11-15:
Since this issue seems solved, we are going to close it. @ryanf2110  Please feel free to open a new issue about multipath processing.