gnss-sdr/gnss-sdr #824: Latency in the PVT solution

Derso1, 2024-11-18:
Hi
I am with a doubt about the latency of the PVT solution.
Is it possible to know what is the latency of PVT solution? 
where I can see this informatiion?

TKS

carlesfernandez, 2024-12-05:
The default GNU Radio buffer in the Observables block is 200 ms (10 observables, 20 ms each). In addition, there is an extra latency in the computation of the PVT solution. However, these numbers do not account for the RF front-end and software driver buffers, which could add some few more milliseconds. Overall, the latency will be about 300-400 ms, depending on your system, and it will not be constant since it will depend on the CPU load.

Derso1, 2024-12-05:
Tks @carlesfernandez 
You helped me a lot.

closed 2024-12-20 by mmajoral