gnss-sdr/gnss-sdr #825: Always loss of lock

slblbwl, 2024-11-29:
I use B210 with a passive antenna to receive L2C signals, but there are often prompts for loss of lock。

[image]

This is my conf file
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2500000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.device_address=
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=5000000
SignalSource.freq=1227600000
SignalSource.clock_source=external
SignalSource.gain=40
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Pass_Through

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=gr_complex

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
Resampler.dump=false
Resampler.dump_filename=../data/resampler.dat
Resampler.item_type=gr_complex
Resampler.sample_freq_in=2500000
Resampler.sample_freq_out=2500000

;######### CHANNELS GLOBAL CONFIG ############
Channels_2S.count=6
Channels.in_acquisition=1

Channel.signal=2S

Channel0.signal=2S
Channel1.signal=2S
Channel2.signal=2S
Channel3.signal=2S
Channel4.signal=2S
Channel5.signal=2S

;######### ACQUISITION GLOBAL CONFIG ############

;# GPS L2C M
Acquisition_2S.implementation=GPS_L2_M_PCPS_Acquisition
Acquisition_2S.item_type=gr_complex
Acquisition_2S.pfa=0.015
;Acquisition_2S.pfa=0.001
Acquisition_2S.doppler_max=3500
Acquisition_2S.doppler_min=-3500
Acquisition_2S.doppler_step=500
Acquisition_2S.max_dwells=1
Acquisition_2S.dump=false
Acquisition_2S.dump_filename=./acq_dump.dat

Tracking_2S.implementation=GPS_L2_M_DLL_PLL_Tracking
Tracking_2S.item_type=gr_complex
Tracking_2S.pll_bw_hz=2.0;
Tracking_2S.dll_bw_hz=0.25;
Tracking_2S.order=2;
Tracking_2S.early_late_space_chips=0.5;
Tracking_2S.dump=false
Tracking_2S.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_2S.implementation=GPS_L2C_Telemetry_Decoder
TelemetryDecoder_2S.dump=false

;######### OBSERVABLES CONFIG ############.
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=PPP_Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=false
PVT.dump_filename=./PVT

Is there a problem with my file or does the antenna need to be replaced? If the antenna needs to be replaced, can you tell me the details of the antenna

arch-dawson, 2024-12-09:
Your sampling rate is pretty low. Increasing that is the first thing that I would try. IIRC L1 and L2 have pretty similar bandwidth, and I normally use 6.25 MHz sampling rate for working with L1.

vladisslav2011, 2024-12-09:
Hello.
`GNSS-SDR.internal_fs_sps=2500000` does not match `SignalSource.sampling_frequency=5000000` and the resampler is disabled `Resampler.implementation=Pass_Through`.
Changing `SignalSource.sampling_frequency` to 2500000 should solve the issue.

mmajoral, 2024-12-20:
Thank you both for your comments and the proposed solution.

closed 2024-12-20 by mmajoral