gnss-sdr/gnss-sdr #826: Many signal conditioners on the same RF channel

joebre, 2024-12-07:
Hi, I just started to use GNSS-SDR for an internal project. 
On a real world GNSS receiver an RF channel topically carries more than one GNNS signal. An example for this is L1 where we often have Beidou B1, GPS/Galileo and Glonass in the same down converted RF band (sig_source output) . In GNSS-SDR this translates into the need of having more "SignalConditioner" connecting to the same RF channel (sig_source output). 

Looking at the examples and at the source code it seems to me that SignalConditioners have an implicit assignment the sig_source output   (signal conditioner 0 connects to sig_source output 0, signal conditioner 1 connects to sig_source output 1, ...  ) ... is this correct ?
If yes, this makes it impossible to connect more signal conditioners to the same signal source output.... Is there any reason for this ?
What would be the best approach to overcome this limitation (possibly avoiding to duplicate channel inside the source)?

Joel

vladisslav2011, 2024-12-07:
Hello.
I've faced the same issue some time ago and successfully resolved it.
It is possible to connect more, than one signal conditioner to the same signal source  since 044bc59.
Here are the config files:
https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/RealTime_input/gnss-sdr_multichannel_GPS_Galileo_Beidou_Glonass_L1_USRP_realtime.conf
https://github.com/gnss-sdr/gnss-sdr/blob/next/conf/RealTime_input/gnss-sdr_multichannel_GPS_Galileo_Beidou_L1_hackrf_realtime.conf

Don't forget to clone and build the 'next' branch.

joebre, 2024-12-08:
Hello Vladislav2011, sorry maybe I used wrongly the name RF channel.
I used the term RF channel  form an GNSS FE HW point of view.

RF channel in the GNSS SDR context has a different meaning with respect to what is typically used in the GNSS HW domain. 

As en example in your .conf  file:
Channels_1B.RF_channel_ID=1 
Channels_1C.RF_channel_ID=1
Channels_B1.RF_channel_ID=0
means connecting the 1B and 1C channels to signal conditioner 1,  and the Beidou B1 channels to  signal conditioner 0, as one can see in gnss_flowgraph.cc - GNSSFlowgraph::connect_signal_conditioners_to_channels(). This is correct only for a 1:1 relationship between the source output channels  and the signal conditioners (1 only signal conditioner for each source channel/HW RF channel).

From a HW point of view, an RF channel consists in an RF down-conversion, filtering and digitization "pipe". 
A GNSS-SDR source topically has 1 channel/stream for each HW RF channel. 

But 1 HW RF channel/source channel is often wide enough to carry several GNSS bands. A typical example for tis case is a wide band  source channel in the L1 band including Beidou B1, GPS/Galielo L1 and GLONASS L1 at different IF frequencies. 
Since GNSS-SDR requires to process  those bands in base band, there is the need to instantiate one signal conditioner for each of these signals with a different IF frequency, all of them connected to the same source channel (ex. channel 0).

It would make sense if gnss_flowgraph.cc - int GNSSFlowgraph::connect_signal_sources_to_signal_conditioners() would loop over the signal conditioners list and connect each signal conditioner to a specified source channel/RF_channel. 

To do this one could add a parameter to the Signal_Conditioner section of the .conf  file which defines what is the source channel/RF channel to which a Signal_Conditioner has to be connected to. As an example on a dual band RF FE (L1/E5ab):

; L1 Band on RF channel/source channel 0
; Signal conditioner for Beidou B1
SignalConditioner0.implementation=Signal_Conditioner
SignalConditioner0.RF_Channel=0

; Signal conditioner for GPS/Galileo 
SignalConditioner1.implementation=Signal_Conditioner
SignalConditioner1.RF_Channel=0

; Signal conditioner for Glonass
SignalConditioner2.implementation=Signal_Conditioner
SignalConditioner2.RF_Channel=0

; E5ab Band on RF channel/source channel 1
; Signal conditioner for GPS/Galileo E5a
SignalConditioner3.implementation=Signal_Conditioner
SignalConditioner3.RF_Channel=1

; Signal conditioner for Galileo E5b
SignalConditioner4.implementation=Signal_Conditioner
SignalConditioner4.RF_Channel=1

Compatibility with alder code could be maintained by assuming the implicit 1:1 relationship between source channels and signal conditioners  when RF_Channel of SignalConditioner<X> has not been defined.

vladisslav2011, 2024-12-08:
> But 1 HW RF channel/source channel is often wide enough to carry several GNSS bands. A typical example for tis case is a wide band source channel in the L1 band including Beidou B1, GPS/Galielo L1 [...]

This exactly what these config files are doing.
B200 config connects 3 signal conditioners with different frequency offsets to receive 4 different systems at 3 different center frequencies extracted from one input stream.
HackRF config connects 2 signal conditioners to receive 3 different systems (Glonass is too far from BeiDou and both does does not fit into ~18MHz) at 2 defferent center frequencies extracted from one input stream.

joebre, 2024-12-08:
Ops sorry .. you're right , AsmoSDR has just 1 source channel and all your conditioners are attached to this single source channel. But what in the case I've multiple source channels ? how so I specify what is the source channel to which a conditioner has to be attached to ?

vladisslav2011, 2024-12-08:
In case of multiple signal sources/channels it would assign channels in order: 1st channel to the 1st conditioner, 2nd channel to the second conditioner and so on.
Overriding the number of channels may work or not work depending on the signal source implementation. For example, it doesn't work with file source, so you'll have to use the `Multichannel_File_Signal_Source` and feed each of it's channels with the same file if you want to extract different system signals from a wideband IF recording.

joebre, 2024-12-08:
Ok, thanks for the explanation ... the situation of having 1 conditioner assigned to each channel was exactly  what I was trying to avoid. It works but it is really sub optimal. 
For the time being, as you're describing above, I'm forced to duplicate the sources in order to have enough channels to "feed" all the conditioners...  it is working but it generates unneeded  CPU load and it is somehow strange from a conceptual point of view.

vladisslav2011, 2024-12-08:
I can agree, that extra channels of  `Multichannel_File_Signal_Source` are generating some unneeded load, but this load is quite low compared to the load from frequency xlating FIR filters (~3% vs 60%). There is no "channel duplication" in case of `Osmosdr_Signal_Source` as it supports "overriding a number of channels", so all signal conditioners are getting connected to the same block. Of course, from conceptual point of view it may be better to have something like `SignalConditioner0.RF_Channel` options to define explicit connections between signal sources and signal conditioners. Feel free to implement it and open a PR.

closed 2026-08-18 by joebre