gnss-sdr/gnss-sdr #828: Carrier phase decimal restricted to small number of values

arch-dawson, 2024-12-09:
I noticed that the decimal portion of the carrier phase outputs are restricted to a specific number of values. To illustrate this, I have constructed plots of the carrier phase outputs modulus 1. Notice that over time, there are fewer decimal values that are represented in the output. The effect seems to worsen with higher carrier doppler shift. Notice especially PRN 10, which has a high Doppler shift. By the end of the run, the carrier phase output decimal portions are restricted to either 0.0 or 0.5. In the best case scenario where the true carrier phase is being rounded to the nearest one of these values, there could be errors of 0.25 rad introduced regularly. These errors are too high for the application that I am using it for.

[image]
[image]
[image]
[image]
[image]

I have looked around in dll_pll_veml_tracking.cc, and the most likely candidate for this type of error seems to be the rounding 
```julia
d_current_prn_length_samples = static_cast<int32_t>(std::floor(d_K_blk_samples));  // round to a discrete number of samples
```
This rounded value is used later when finding the accumulated carrier phase. 
```julia
d_acc_carrier_phase_rad -= (d_carrier_phase_step_rad * static_cast<double>(d_current_prn_length_samples) + 0.5 * d_carrier_phase_rate_step_rad * static_cast<double>(d_current_prn_length_samples) * static_cast<double>(d_current_prn_length_samples));
```

Miscellaneous other facts:
- I have restricted what is plotted here to "state 4", as used in dll_pll_veml_tracking.cc 
- The receiver is static during this test sitting in the middle of a field. Nothing unusual was going on with my setup, as best I can tell
- The oscillator that I am using is an OCXO
- The Doppler shift value listed in the figure title is the average value over the plotting duration

My configuration file:
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=3125000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=<redacted>
SignalSource.item_type=ishort
SignalSource.sampling_frequency=6250000
SignalSource.samples=0
SignalSource.seconds_to_skip=120

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Ishort_To_Cshort

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Fir_Filter
InputFilter.input_item_type=cshort
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=11
InputFilter.number_of_bands=2

InputFilter.band1_begin=0.0
InputFilter.band1_end=0.48
InputFilter.band2_begin=0.52
InputFilter.band2_end=1.0

InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0

InputFilter.band1_error=1.0
InputFilter.band2_error=1.0

InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=5000000
InputFilter.IF=0

;######### RESAMPLER CONFIG ############
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=6250000
Resampler.sample_freq_out=3125000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=5
Channel0.satellite=2
Channel1.satellite=10
Channel2.satellite=21
Channel3.satellite=24
Channel4.satellite=32

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=10
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=<redacted>
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.bit_transition_flag=false

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.enable_fll_pull_in=false
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=true
Tracking_1C.dump_mat=true
Tracking_1C.dump_filename=<redacted>
Tracking_1C.carrier_aiding=true

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_mat=true
Observables.dump_filename=<redacted>

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.output_enabled=false
PVT.gpx_output_enabled=false
PVT.dump=true
PVT.dump_mat=true
PVT.dump_filename=<redacted>
```

mmajoral, 2024-12-20:
Thank you for your comment, we will try to reproduce this phenomenon

mmajoral, 2024-12-20:
Thank you for your comments and for sharing this information. Congratulations on your outstanding work!

arch-dawson, 2025-02-10:
Have you been able to reproduce this? I found that the effect is slightly worse than I thought. I looked at a couple of longer (15 ish minute) data sets and for a few of the PRNs at the end all of the carrier phase values mod2 were zero. i.e. the outputs were rounded to the nearest even number. I love using gnss-sdr, but that's a crazy amount of error to have in carrier phase.

arch-dawson, 2025-02-10:
I just figured out the problem, actually. The carrier phase is being stored as a float32 and it's running out of precision. The effect will occur more quickly with higher doppler rates because the value reaches the thresholds for loss of precision more quickly. You should be able to fix this error by switching carrier phase to a float64.

arch-dawson, 2025-02-10:
I found a fix for this issue. The carrier phase value in the tracking loop is stored as a double, but it is only dumped as a float, so a lot of precision is lost. As the carrier phase value gets larger and crosses different thresholds (e.g. passes 2^23) then precision in the decimal place will be lost, and this odd rounding behavior appears. The simple fix for this issue is outputting the acc_carrier_phase_rad variable as a double rather than as a float.

Several lines need to be changed in dll_pll_veml_tracking.cc to fix the issue; I included the relevant modified lines below. I don't see any downside to this change beyond slightly larger file size. If the maintainer's agree that this is a reasonable change, I am happy to make a pull request changing the dll_pll_veml_tracking.cc file and the others where this issue is likely to occur. I can also check the other variables that are dumped to make sure that we can reasonably expect a double to properly store the required precision. 

```c++
tmp_double = static_cast<double>(d_acc_carrier_phase_rad);
d_dump_file.write(reinterpret_cast<char *>(&tmp_double), sizeof(double));
...
const int32_t number_of_double_vars = 2;
const int32_t number_of_float_vars = 18;
...
auto acc_carrier_phase_rad = std::vector<double>(num_epoch);
...
dump_file.read(reinterpret_cast<char *>(&acc_carrier_phase_rad[i]), sizeof(double));
...
matvar = Mat_VarCreate("acc_carrier_phase_rad", MAT_C_DOUBLE, MAT_T_DOUBLE, 2, dims.data(), acc_carrier_phase_rad.data(), 0);
```