gnss-sdr/gnss-sdr #829: build of v0.0.19 fails: there is error report after using "make" command

Rango-T10000, 2024-12-10:
I am using Ubuntu 22.04 + gnuradio 3.10.7.0
I have installed all the packet according to your instruction.
And when I try to build using "make" I met the following error:
[image]

Rango-T10000, 2024-12-10:
I guess this error is about libfmt-dev libspdlog-dev.
And I also find there are issures talking about this.
But none of them provide the solution.

jwmelto, 2024-12-10:
The error says `basic_runtime` is not known in the `fmt` namespace. Are you using an external **fmtlib** or the **spdlog** built-in one? I don't see any **fmt** include files in your trace. 

I had trouble building only part of the chain from source, so I built GNU Radio from source as well. I also built **fmtlib** and **spdlog** from source (not necessarily for this reason).

Rango-T10000, 2024-12-11:
> The error says `basic_runtime` is not known in the `fmt` namespace. Are you using an external **fmtlib** or the **spdlog** built-in one? I don't see any **fmt** include files in your trace.
>
> I had [...]

Thanks for your reply. When I install all the package by :
`$ sudo apt-get install build-essential cmake git pkg-config libboost-dev libboost-date-time-dev \
       libboost-system-dev libboost-filesystem-dev libboost-thread-dev libboost-chrono-dev \
       libboost-serialization-dev liblog4cpp5-dev libuhd-dev gnuradio-dev gr-osmosdr \
       libblas-dev liblapack-dev libarmadillo-dev libgflags-dev libgoogle-glog-dev \
       libgnutls-openssl-dev libpcap-dev libmatio-dev libpugixml-dev libgtest-dev \
       libprotobuf-dev protobuf-compiler python3-mako`
       
 It shoews:
[image]

Do you mean intall fmtlib and spdlog from source can slove this problem?

Rango-T10000, 2024-12-11:
I think I slove this problem. I build and install GNSS_SDR 0.0.19 from source code successfully.
[image]

I want to make a summary here for people who have the same problem.  The following is what I have done:

1. According to the instruction: I install all dependencies using software packages buy command:
`sudo apt-get install build-essential cmake git pkg-config libboost-dev libboost-date-time-dev \
       libboost-system-dev libboost-filesystem-dev libboost-thread-dev libboost-chrono-dev \
       libboost-serialization-dev liblog4cpp5-dev libuhd-dev gnuradio-dev gr-osmosdr \
       libblas-dev liblapack-dev libarmadillo-dev libgflags-dev libgoogle-glog-dev \
       libgnutls-openssl-dev libpcap-dev libmatio-dev libpugixml-dev libgtest-dev \
       libprotobuf-dev protobuf-compiler python3-mako`
       
2. Then I follow the regular steps:
`git clone https://github.com/gnss-sdr/gnss-sdr`
`$ cd gnss-sdr/build`
`cmake ..`
'make'

However, I met the error here that libfmt and libspdlog are incomplatible!
According to https://github.com/gabime/spdlog/issues/2614#issuecomment-1396348189,
it seems spdlog v1.11.0 uses fmt v9.1.0, so I need to downgrade the fmt to v9.1.0.
(It needs to install fmt v9.1.0 from source code; I am not sure if spdlog needs the same operation, you can try if only install fmt v9.1.0 can make gnss-sdr's build successfully)

The steps of installing fmt from source code are:
`git clone --branch 9.1.0 https://github.com/fmtlib/fmt.git
cd fmt
`
`mkdir build && cd build
cmake .. 
make -j$(nproc)
sudo make install
`

3. After installing the fmt v9.1.0, you can build GNSS_SDR aging:
you can delet the "build" folder of GNSS_SDR by `rm -rf build`
And then follow the regular steps again:
`mkdir build && cd build
cmake .. 
make -j$(nproc)
sudo make install
`

And If everying works fine, try `gnss-sdr --version`.
If it shows :"gnss_sdr version 0.0.19", I think the build and install are successful.

mmajoral, 2024-12-20:
We recommend using the 'next' branch, as the documentation is always up-to-date for that branch.

closed 2024-12-20 by mmajoral

Rango-T10000, 2024-12-23:
Besides I install this gnss-sdr in my Ubuntu22.04, I also install it in my MacOS15.2  and I also met a lot of problems!

1. According to the instruction: I install the Xcode from Appstore at first.
[image]

2. According to the "Software pre-requisites can be installed using either [Macports](https://github.com/gnss-sdr/gnss-sdr?tab=readme-ov-file#macports) or [Homebrew](https://github.com/gnss-sdr/gnss-sdr?tab=readme-ov-file#homebrew)." And I have tried both Macports and Homebrew. Before using these two tools to install the required dependencies, you need to install the tools(Macports or Homebrew) at first. And then, install the required dependencies. Using Macports is very slow!

3. After installing all the dependencies, you can build gnss-sdr from source code now. The typical steps are:
```
git clone https://github.com/gnss-sdr/gnss-sdr
cd gnss-sdr/build
cmake ..
make
sudo make install
```
And not surprisingly, you will met ao many problem when using `cmake..` or `make`. All the stupid problems are about the following two situations:
【1】Some required dependencies are not installed------------->try to install this packet using tools(Macports or Homebrew)
【2】Some required dependencies are installed, but the versions are not compatible!----------------->try to install another packet using tools(Macports or Homebrew) or just build it from source code.

Besides, if error happens after using `cmake..`, you should using the following command to restart:
```
rm -rf build (or if your are in build folder, you can just rm -rf *)
mkdir build
cd build
cmake ..
```
Besides, if error happens after using `make`, you should using the following command to restart:
```
make clean
make
```
4. When I build it in my MacOS, I met the following problems:
[1] The version of `boost` is not compatible!
[image]
[image]

[2] Problem about `gtest` packet:
[image]
[image]
[image]

5. Finally I install the gnss-sdr successfully!
After using `sudo make install`, remember to go to the gnss-sdr/install folder, and run `volk_profile` and `volk_gnsssdr_profile`.
[image]

Rango-T10000, 2024-12-25:
> We recommend using the 'next' branch, as the documentation is always up-to-date for that branch.

@mmajoral, I also try to install on my mac. I can cmake and make successfully.
However, I still met problem that I can not use gnss-sdr.
The details are in https://github.com/gnss-sdr/gnss-sdr/issues/835#issuecomment-2560897806
Could you please give me some advice?