gnss-sdr/gnss-sdr #831: Cannot receive the GNSS signal (URSP X310, active GNSS antenna, bias-T)

Rango-T10000, 2024-12-14:
My device is USRP X310(two TwinRX-80MHz daughterboards), active GNSS antenna(3~5 V), bias-T for power the antenna.
[image]
[image]

**Testbed Testing**
1. I use another USRP X310 with passive GPS antenna to transmit sine signal at 1.575GHz, and I can receive this signal with my testbed discribed above. I can clearly see the signal hump with the GUNRadio GUI's frequency sink. As a result, I am sure my testbed discribed above can work. And every part works good.

Here are my GUNRadio GUI's setup:
[image]

**Post-processing mode**
2. I go to the outdoor area and try to receive the real GPS signal. First of all, I try to receive it in a post-processing mode, which means I use GUNRadio GUI to record the raw data and keep it as a .dat file. However, I can't see any signal hump with the GUNRadio GUI's frequency sink. Even I have set the rx_gain as 90, I can see anything. As a result, I am not sure if I receive the signal or not. After record these 4 .dat files, I try to run gnss-sdr to decode the .dat file. However, it failed. I can't get the output files.
[image]
[image]

**Real-time receiving mode**
3. Then I try to receive the GPS signal in a real-time receiving mode with just one antenna. However, it still failed.I can't get any output files.
[image]

carlesfernandez, 2024-12-14:
`GNSS-SDR.internal_fs_sps=2000000`, since you are resampling the input signal.

Rango-T10000, 2024-12-14:
> `GNSS-SDR.internal_fs_sps=2000000`, since you are resampling the input signal.

I copy it from your config example.
So should I just comment this "GNSS-SDR.internal_fs_sps=2000000"?
And also set the Resampler.implementation=Pass_Through to disable the resampler?

Rango-T10000, 2024-12-14:
> `GNSS-SDR.internal_fs_sps=2000000`, since you are resampling the input signal.

This is my modified config for post-processing:
`[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/ddh/ssd2/gnss-sdr/fsc_exp/data/ant0_20241214_160903344507.dat
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=4000000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
# DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
#Resampler.implementation=Direct_Resampler
Resampler.implementation=Pass_Through
# Resampler.sample_freq_in=4000000
# Resampler.sample_freq_out=2000000
# Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=8
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250
Acquisition_1C.blocking=true

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=3
`

Could you please tell me which part is wrong?
Because from your document, this gnss-sdr includes several parts(Signal Source, signal conditioner[input filter, resampler], channels, observables, PVT....). I don't understande which part is needed. In your provided config examples, there are too many versions and I don't know which one is the most suitble for my setup.

Rango-T10000, 2024-12-15:
https://github.com/gnss-sdr/gnss-sdr/issues/421
https://github.com/gnss-sdr/gnss-sdr/issues/376

Rango-T10000, 2024-12-16:
I modify my testbed by add an external GPSDO:
[image]
[image]

**Real-time receive mode**
1. I use the following config file and try to receive the signal with only one antenna and in a real-time processing mode. It shows some GPS NEV message received, but still with no right output files.
[image]
[image]
[image]

Rango-T10000, 2024-12-16:
About above discribed overflow error:
"2. I use the Gnuradio GUI to receive the raw data at first. However, I met the error: overflow occurred!
The sample rate is 4MSa/s for each RF source(I have 4 RF source, i.e. 4 ants), the data format is gr_complex(complex 32float for I and Q). As a result the total data generated is 4MSa/s * 4 ants * 4B * 2 = 128MB/s, which I think doesn't exceed the bandwidth between PC and USRP."

I use SFP+ (Small Form-factor Pluggable Plus) in USRP X310 to communicate with my PC which should reach 10Gbps(1000Mbps) which is 1250MB/s. 
[image]

However, the Ethernet controller(网卡) in my PC has only 1000Mb/s(1 Gbps) speed which is 125MB/s! As aresult, my requiremt is 128MB/s which is bigger than 125MB/s.
[image]

My solution is to use a 10Gbps Ethernet controller(Thunderbolt™ 4 port to 10Gbps Ethernet port).

Rango-T10000, 2024-12-17:
I use only one antenna in a real-time receiving mode to receive the successfully. 
However, it's quite unstable, still loss of lock usually.

About external "GPSDO", it needs 3~5min to lock and run correctly. When it runs correctly, the red LED(ALM) should extinguish，the GPS lock LED should be lighting, and the RUN LED should blink. Like this:
 [image]

And after using `uhd_usrp_probe` command, it shows:
[image]

With the following config file, I only received and decoded the signal once.
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000 ; Sampling frequency in [Hz]
SignalSource.freq=1575420000 ; RF front-end center frequency in [Hz]
SignalSource.gain=70 ; Front-end gain in dB
SignalSource.subdevice=A:0 ; UHD subdevice specification (for USRP1 use A:0 or B:0, for USRP B210 use A:0)
SignalSource.clock=external ;Use GPSDO as the external clock: 10MHz -> REF IN
SignalSource.time=external;  Use GPSDO as the external time: 1PPS -> PPS IN

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
# DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Pass_Through
# Resampler.implementation=Direct_Resampler
# Resampler.sample_freq_in=4000000
# Resampler.sample_freq_out=2000000
# Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=8
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250
Acquisition_1C.blocking=true

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=3
```
[image]

It is still very unstable, could anyone give me some advice to improve?

mmajoral, 2024-12-20:
Thank you for your comments and for sharing this information. Congratulations on your outstanding work!

closed 2024-12-20 by mmajoral

Rango-T10000, 2024-12-28:
If you have interest in antenna array receiver(1 x4 antenna array), please check https://github.com/Rango-T10000/Antenna_array_GPS_receiver_USRP_GNSS-SDR.

carolim-git, 2026-01-14:
Hi, 
Thank you for sharing your work ! 
I have a question regarding your configuration : SignalSource.time=external;
I would like to use my USRP Ettus X310 with an external 1 PPS signal (not gpsdo) as PPS_in, I think I should add SignalSource.time=external; in my configuration, but I am not sure
Can you confirm ?
Thank you
Caroline