gnss-sdr/gnss-sdr #835: Installation difficulties on OSX 15.2 through Homebrew

aWo-git, 2024-12-24:
Hey Everyone, 
since I'm not at all a programmer I ran into several difficulties while installing gnss-sdr. 

### The first problem context was mako:
1. I could only install mako through `pipx install mako` not `pip install mako` as described in the readme.md 
2. Pipx installs mako in a virtual environment which (I think) is the real problem here, since mako could not be found in the `cmake ..` process. 
3. After a lot of trying stuff to make mako available during `cmake ..` (e.g. including the module path to the $PATH Variable, the $PYTHONPATH Variable, activating the venv at startup from the terminal I ended up with including the following (Replace YOURUSR with your real username. 
`import sys
 sys.path.insert(0,'/Users/YOURUSR/.local/pipx/venvs/mako/lib/python3.13/site-packages')`
in the files
`./gnss-sdr/cmake/Modules/SetupPython.cmake` 
`./gnss-sdr/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/cmake/Modules/VolkPython.cmake`
in the XXX_PYTHON_CHECK_MODULE (replace XXX with GNSSSDR or VOLK) Macros after the first `try`:
`execute_process(`
`    COMMAND ${PYTHON_EXECUTABLE} -c "`
`#########################################`  
`try:`   
`    import sys`  
`    sys.path.insert(0,'/Users/YOURUSR/.local/pipx/venvs/mako/lib/python3.13/site-packages')`  
`    import ${mod}`

and also included the `sys.path.insert(0,'/Users/YOURUSR/.local/pipx/venvs/mako/lib/python3.13/site-packages')` in the file
`./gnss-sdr/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/gen/volk_gnsssdr_tmpl_utils.py`
at line 18 before the call of mako
`sys.path.insert(0,'/Users/aWo/.local/pipx/venvs/mako/lib/python3.13/site-packages')`
`from mako.template import Template`

### The second problem context was Boost
1. Installed boost through homebrew to `/opt/homebrew/Cellar/boost@1.85`
2. During `make` the `format.hpp` could not be found: `/opt/homebrew/Cellar/gnuradio/3.10.11.0_3/include/gnuradio/logger.h:47:10: fatal error: 'boost/format.hpp' file not found`
`47 | #include <boost/format.hpp>`
3. to get around this problem I added the following to the file `./gnss-sdr/src/algorithms/conditioner/adapters/CMakeLists.txt`: 
at line 48 in the `target_include_directories` 
`target_include_directories(conditioner_adapters`
`    PUBLIC`
`       ${GNSSSDR_SOURCE_DIR}/src/core/interfaces`
`        /opt/homebrew/Cellar/boost@1.85/1.85.0_2/include`
`)

### Conclusions
After this additions the sw installs smoothly, but there should be a better way. Perhaps somebody more knowledgeable could make something out of my trials and errors and include changes to the install process for OSX with homebrew. 

Thanks for the SW, since it is now running, I can play with it in the next days

Rango-T10000, 2024-12-24:
After testing a lot of version, I found `boost`1.81 may be suitble.
[image]

However, you can't install `boost` 1.81 via Macport or brew. The only way is to download and build it from the packet provided by [https://www.boost.org/users/history/](url).

The detailed steps are:
```
wget https://boostorg.jfrog.io/artifactory/main/release/1.81.0/source/boost_1_81_0.tar.gz
tar -xvzf boost_1_81_0.tar.gz
cd boost_1_81_0
sudo ./bootstrap.sh --with-toolset=clang
sudo ./b2 headers   
./b2 install --prefix=/usr/local/boost_1_81_0
```

After that, try to build gnss-sdr again by:
```
cmake ..
make
```

mmajoral, 2025-02-06:
Thank you for your comments. As stated above we recommend using the "next" branch of GNSS-SDR to try out the latest developments.

closed 2025-02-06 by mmajoral