gnss-sdr/gnss-sdr #837: GNSS-SDR.assist_dual_frequency_acq is not working for Galileo_E5a_Noncoherent_IQ_Acquisition_CAF

joebre, 2025-01-02:
The acquisition Doppler aiding mechanism is not working for Galileo_E5a_Noncoherent_IQ_Acquisition_CAF. 
The search Doppler center frequency is always set to zero even when  GNSS-SDR.assist_dual_frequency_acq is set to true. 

Using  Galileo_E5a_Pcps_Acquisition the frequency aiding is  properly computed and used as acquisition center frequency.

joebre, 2025-01-03:
Taking a look to the code it seems  that the reason for which the Doppler aiding is not working is that Galileo_E5a_Noncoherent_IQ_Acquisition_CAF do not  implements set_doppler_center() and can't therefore take advantage of the frequency aiding provided by tracked E1 signals.
I'll replace Galileo_E5a_Noncoherent_IQ_Acquisition_CAF with the more complete Galileo_E5a_Pcps_Acquisition...
Joel

mmajoral, 2025-02-06:
Thank you for reporting this issue. We will have a look into it.