gnss-sdr/gnss-sdr #844: No sync from GPS data file

ZimBloo, 2025-01-14:
Hi, I have been given the task of decoding a GPS L1 (1575.42MHz) signal in real time. I could not find anywhere how to create this using GNU-RADIO. I figured I would store a GPS data file and use the GNSS-SDR software to decode it. I am using HackRF One with osmocom source block and file sink. But I got to the point where gnss, after running the configuration file (copied from https://gnss-sdr.org/my-first-fix/), all it says is "Tracking of GPS L1 C/A signal ....." for channels 0-7. It then prints Loss of lock and a moment later again "Tracking of GPS L1 C/A signal .....". There is no synchronisation. I have set samp_freq to 4M, using gr_complex units in GNSS (Complex Float 32 in GNU). Do you have any advice on how to solve this problem? I am new to using GNU and GNSS-SDR. Any advice will be gratefully received.

arch-dawson, 2025-01-18:
How confident are you in your hardware setup? Do you have a good view of the sky? Powered GPS antenna?

ZimBloo, 2025-01-23:
Sorry for the late reply, I was very busy. The antenna is placed outside with a direct view of the sky. The GPS antenna is powered by the SDR.
In the meantime, I tried decoding using only GNSS-SDR and followed the steps https://gnss-sdr.org/conf/#gps-l1-ca-receiver-using-a-hackrf-one. I currently have identical results as with the above problem. An external clock with a frequency stability of 2ppm is connected to the SDR. I don't know where to look for the problem because it shows me Current receiver time: 1s .... 12s, then Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 23 (Block III) and nothing for another 10s until Loss of lock in channel 1.

mmajoral, 2025-02-06:
It is difficult to say what the problem is from the information we have. We suggest you verify the HackRF is producing the output bias voltage to power up the antenna, it can be checked with a simple multimeter at the SMA antenna input. In addition, it might be helpful to check that HackRF is effectively detecting the external clock. There is a command line utility related to the HackRF software to do that.

YyyingT, 2025-04-02:
@ZimBloo I've encountered the same issue as well, where it often shows 'loss of track' after tracking. How did you improve it later on? Thank you!