gnss-sdr/gnss-sdr #846: Duplicated and irregular Observation sets passed to rtkpos

joebre, 2025-02-04:
Hello, I was trying to identify the cause of the issue which makes GNSS-SDR next unable to perform continuous fix in multi frequency configuration (E1 + E5a) ... For doing this I instrumented a little rtklib_solver.cc.  I added:

std::cout << "Observation Set\n";
for (int i = 0; i < valid_obs; i++)
        {                                   
            std::cout  << "\tPRN: "<< (int)d_obs_data[i].sat << // << "T: "<< lstr <<
                "\tTime: "<< d_obs_data[i].time.time <<
                "\tSec: "<< d_obs_data[i].time.sec << 
                "\tL[0]: "<< d_obs_data[i].L[0] << std::endl;
        }

just after the  after the d_obs_data population loop. 

This code  shows me with which receiver time the observation in d_obs_data are passed to rtkpos().

Using a GNSS-SDR.observable_interval_ms = 200, after a while,  I got the following output:

_Observation Set
	PRN: 64	Time: 1734959137	Sec: 0.600000	L[0]: 32823.390308
	PRN: 62	Time: 1734959137	Sec: 0.600000	L[0]: -12782.479808
	PRN: 90	Time: 1734959137	Sec: 0.600000	L[0]: 36797.156987
	PRN: 83	Time: 1734959137	Sec: 0.600000	L[0]: 15623.965566
	PRN: 71	Time: 1734959137	Sec: 0.600000	L[0]: -29795.949749
	PRN: 84	Time: 1734959137	Sec: 0.600000	L[0]: -23744.605468
Observation Set
	PRN: 64	Time: 1734959137	Sec: 0.800000	L[0]: 33230.721248
	PRN: 62	Time: 1734959137	Sec: 0.800000	L[0]: -12936.347384
	PRN: 90	Time: 1734959137	Sec: 0.800000	L[0]: 37293.543229
	PRN: 83	Time: 1734959137	Sec: 0.800000	L[0]: 15814.432946
	PRN: 71	Time: 1734959137	Sec: 0.800000	L[0]: -30308.449903
	PRN: 84	Time: 1734959137	Sec: 0.800000	L[0]: -24036.733517
Observation Set
	PRN: 64	Time: 1734959138	Sec: 0.000000	L[0]: 33638.052040
	PRN: 62	Time: 1734959138	Sec: 0.000000	L[0]: -13090.188565
	PRN: 90	Time: 1734959138	Sec: 0.000000	L[0]: 37789.917655
	PRN: 83	Time: 1734959138	Sec: 0.000000	L[0]: 16004.902380
	PRN: 71	Time: 1734959138	Sec: 0.000000	L[0]: -30820.961575
	PRN: 84	Time: 1734959138	Sec: 0.000000	L[0]: -24328.861294
Observation Set
	PRN: 64	Time: 1734959138	Sec: 0.000000	L[0]: 33638.052040
	PRN: 62	Time: 1734959138	Sec: 0.000000	L[0]: -13090.188565
	PRN: 90	Time: 1734959138	Sec: 0.000000	L[0]: 37789.917655
	PRN: 83	Time: 1734959138	Sec: 0.000000	L[0]: 16004.902380
	PRN: 71	Time: 1734959138	Sec: 0.000000	L[0]: -30820.961575
	PRN: 84	Time: 1734959138	Sec: 0.000000	L[0]: -24328.861294
Observation Set
	PRN: 64	Time: 1734959138	Sec: 0.200000	L[0]: 34045.393919
	PRN: 62	Time: 1734959138	Sec: 0.200000	L[0]: -13244.028436
	PRN: 90	Time: 1734959138	Sec: 0.200000	L[0]: 38286.304343
	PRN: 83	Time: 1734959138	Sec: 0.200000	L[0]: 16195.373337
	PRN: 71	Time: 1734959138	Sec: 0.200000	L[0]: -31333.447842
	PRN: 84	Time: 1734959138	Sec: 0.200000	L[0]: -24620.988206
	PRN: 92	Time: 1734959138	Sec: 0.200000	L[0]: -8400.180914_

The observation set of  Time: 1734959138	Sec: 0.000000  is submitted twice to rtkpos.  Whatever the observable_interval_ms is the observation set on the second boundary gets submitted twice.. 

Is  this an expected behavior ? If yes, why ? Bug ?
Thx, Joel

MathieuFavreau, 2025-02-21:
There are two rtklib_solver instances as you can find in `rtklib_pvt_gs.h`:
```
std::shared_ptr<Rtklib_Solver> d_internal_pvt_solver;
std::shared_ptr<Rtklib_Solver> d_user_pvt_solver;
```
The second one is used for the RINEX OBS logging for example, and it's calculation rate is dictated by:
```
PVT.display_rate_ms
```

The default value is 500ms, which explains why you only get twice the logs every seconds (200ms and 500ms only intersects every seconds)

joebre, 2025-02-27:
Ops... Ok .. I see... many thanks !

closed 2026-08-18 by joebre