gnss-sdr/gnss-sdr #848: Building: `cmake` call does not seem to terminate and uses a lot of CPU.

dreirund, 2025-02-14:
Ahoj,

I am trying to build this (git checkout from master branch; current git commit hash: a6daa5ad7) on [Artix GNU/Linux](http://artixlinux.org/) (rolling release).

* cmake version: 3.31.5,
* GCC version: 14.2.1.

`cmake` configuration hangs without terminating (for at least 30 minutes) at high CPU usage and high disk read.

I run `cmake` as follows:  
```bash
cmake \
  --debug-output \
  --debug-find \
  --trace-expand \
  -S gnss-sdr \
  -B build \
  -DCMAKE_BUILD_TYPE=Release \
  -DCMAKE_INSTALL_PREFIX=/usr \
  -DENABLE_STRIP=ON \
  -DENABLE_SYSTEM_TESTING=OFF \
  -DENABLE_SYSTEM_TESTING_EXTRA=OFF \
  -DENABLE_UNIT_TESTING=OFF \
  -DENABLE_UNIT_TESTING_EXTRA=OFF \
  -DENABLE_UNIT_TESTING_MINIMAL=OFF \
  -DENABLE_INSTALL_TESTS=OFF \
  -DBoost_USE_STATIC_LIBS=OFF \
  -DENABLE_AD9361=ON \
  -DENABLE_AD936X_SDR=ON \
  -DENABLE_ARRAY=ON \
  -DENABLE_BENCHMARKS=OFF \
  -DENABLE_CPUFEATURES=ON \
  -DENABLE_CUDA=OFF \
  -DENABLE_DMA_PROXY=OFF \
  -DENABLE_EXTERNAL_MATHJAX=OFF \
  -DENABLE_FLEXIBAND=ON \
  -DENABLE_FMCOMMS2=ON \
  -DENABLE_FPGA=ON \
  -DENABLE_GNSS_SIM_INSTALL=ON \
  -DENABLE_GNUTLS=ON \
  -DENABLE_ION=ON \
  -DENABLE_LIMESDR=ON \
  -DENABLE_LOG=ON \
  -DENABLE_MAX2771=ON \
  -DENABLE_OPENCL=OFF \
  -DENABLE_ORC=ON \
  -DENABLE_OSMOSDR=ON \
  -DENABLE_OWN_ABSEIL=OFF \
  -DENABLE_OWN_ARMADILLO=OFF \
  -DENABLE_OWN_GLOG=OFF \
  -DENABLE_OWN_GNSSTK=OFF \
  -DENABLE_OWN_GPSTK=OFF \
  -DENABLE_PACKAGING=OFF \
  -DENABLE_PLUTOSDR=ON \
  -DENABLE_PROFILING=OFF \
  -DENABLE_RAW_UDP=ON \
  -DENABLE_UHD=ON \
  -DENABLE_ZMQ=ON
```

The last lines of this `cmake` trace output are:  
```
[...]
/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt(3364):  set_target_properties(ION::ion PROPERTIES IMPORTED_LINK_INTERFACE_LANGUAGES CXX IMPORTED_LOCATION /var/cache/makepkg/build/gnss-sdr-git/src/build/gnss-metadata-standard/source/api/lib/GnssMetadata/libapi.a INTERFACE_INCLUDE_DIRECTORIES /var/cache/makepkg/build/gnss-sdr-git/src/build/thirdparty/gnss-metadata-standard/source/api/inc INTERFACE_LINK_LIBRARIES /var/cache/makepkg/build/gnss-sdr-git/src/build/gnss-metadata-standard/source/api/lib/GnssMetadata/libapi.a;/var/cache/makepkg/build/gnss-sdr-git/src/build/gnss-metadata-standard/source/api/lib/tinyxml2/libxml.a )
   Called from: [1]	/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt
/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt(3379):  if(ENABLE_MAX2771 AND NOT ENABLE_FPGA )
   Called from: [1]	/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt
/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt(3387):  if(ENABLE_DMA_PROXY AND NOT ENABLE_FPGA )
   Called from: [1]	/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt
/var/cache/makepkg/build/gnss-sdr-git/src/gnss-sdr/CMakeLists.txt(3403):  if(ENABLE_MAX2771 )
[... 231 lines cut ...]
-- Found GRLIMESDR: /usr/lib/libgnuradio-limesdr.so
-- The (optional) gr-limesdr module has been found.
--  Using standard gr-limesdr library.
-- The driver for LimeSDR will be compiled.
--  You can disable it with 'cmake -DENABLE_LIMESDR=OFF ..'
-- Checking for module 'libad9361'
--   Found libad9361, version 0.2
-- Found LIBAD9361: /usr/lib/libad9361.so
-- Checking for module 'libiio'
--   Found libiio, version 0.26
-- Found LIBIIO: /usr/lib/libiio.so
CMake Deprecation Warning at /var/cache/makepkg/build/gnss-sdr-git/src/build/thirdparty/gnss-metadata-standard/CMakeLists.txt:32 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  -std=c++11 -Wall -Wextra -pedantic
```

Also attached: [`cmake.log`](https://github.com/user-attachments/files/18791861/cmake.log).

Regards!

jwmelto, 2025-02-14:
CMake 3.31 is pretty new, and **master** is pretty old. Try building **next**, which is the current development branch.

dreirund, 2025-02-15:
> *CMake 3.31 is pretty new, and **master** is pretty old. Try building **next**, which is the current development branch.*

I actually use the [`next` branch](https://github.com/gnss-sdr/gnss-sdr/tree/next).

[Commit `a6daa5a`](https://github.com/gnss-sdr/gnss-sdr/commit/a6daa5ad7e94a2ee910fe0dff980c8d3bf05c7f6) actually was commited on 2025-02-13 and is in the [`next` branch](https://github.com/gnss-sdr/gnss-sdr/tree/next).

dreirund, 2025-03-29:
Hm it was triggered by some combination of options which actually required dependencies I did not have.

But instead of complaining, `cmake` got stuck.

Regards!

closed 2025-03-29 by dreirund