gnss-sdr/gnss-sdr #849: Tracking_sample_counter value in hybrid_observables.cc

emingunes1, 2025-02-14:
In the context of position estimation, the value named Tracking_sample_counter in the hybrid_observables.cc file is used for calculating time and position. This value seems to increase by 52,000 samples at each interval (e.g., in the 50 Hz – 20 ms operating range and for 2.6 MHz sampling). However, the value doesn’t come as an exact multiple of 26,000; instead, it shows up as a slightly offset number.

To explain better, the values obtained every 20 ms are as follows:
323556
349556
375556
401556

If the number of samples per millisecond were being counted exactly, these values should have been exact multiples of 26,000. However, the observed values look more like (x + 26000 * loop_count).

Honestly, I couldn’t figure out where this x value is coming from.

I’m trying to find the theoretical meaning of the value x in a receiver algorithm.

Sorry if the question is a bit complicated, but this was the best way I could explain it.

arch-dawson, 2025-02-15:
The receiver has to pick times when it should produce a navigation solution. Let's consider a few possible design choices:

Option A: The receiver produces a solution every 0.01 seconds of measured receiver time. So at integer multiples of 26,000 samples, the receiver will produce a batch of measurements. In this case x would be zero.

Option B: The receiver waits until it is tracking >4 signals, and then produces measurements every 0.01 seconds after that. So if, starting at sample 323556 the receiver locks on to its fourth signal, it will produce measurements for all of the PRNs at sample 323556, then wait 0.01 seconds and produce another at sample 349556, etc. In this case x would be the number of samples until the receiver has locked on to four signals. 

Option C: The receiver produces observables at data bit times. The data bit times are arbitrary from the perspective of the receiver. It will take the receiver a short time to track the signal and find the data bit transitions, and then it can start producing observables after that. In this case x would be the number of samples to the first data bit that the receiver identifies. 

It sounds like you are expecting the receiver to use Option A, but that is not the case. GNSS-SDR must be using Option B, C, or some other option not listed above. Any of the design choices are reasonable. I'm guessing that the designers chose Option C, but you can check by comparing the observable times against the tracking output times when the tracking is in stage 4.

emingunes1, 2025-02-17:
Thank you. Actually, at first, we also thought it was related to data bit synchronization. However, since the data bit comes every 20 ms and the number of samples per millisecond is known, it doesn’t seem like the sample_counter value being in between is related to bit synchronization. This is because there’s no need for sample-level precision when extracting the bits. Tracking with 1 ms precision is sufficient. Could the x value I mentioned be related to acquisition or the code phase present there?

arch-dawson, 2025-02-18:
I don't understand what this part of your message means:

> However, since the data bit comes every 20 ms and the number of samples per millisecond is known, it doesn’t seem like the sample_counter value being in between is related to bit synchronization. [...]

Let's assume that Option C is correct and GNSS-SDR does produce observables at data bit times. The software tracks the location in the file by sample number, so it outputs the sample number of the start of the data bit when it outputs an observable. I don't understand what you mean when you say that tracking with 1ms precision is sufficient.