gnss-sdr/gnss-sdr #852: `volk_gnsssdr_module` installation failure?

doronbehar, 2025-02-20:
So on NixOS, ever since a while ago we started experiencing the following installation failure:

```
Install the project...
-- Install configuration: "Release"
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/pkgconfig/volk_gnsssdr.pc
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8ic_magnitude_squared_8i.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_x2_rotator_dot_prod_32fc_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32f_high_dynamics_resamplerxnpuppet_32f.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32f_xn_high_dynamics_resampler_32f_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8ic_x2_dot_prod_8ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_32f_rotator_dotprodxnpuppet_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32f_xn_resampler_32f_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_s32f_sincos_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_x2_dot_prod_16ic_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_resamplerxnpuppet_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16i_resamplerxnpuppet_16i.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_32f_rotator_dot_prod_32fc_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_convert_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_conjugate_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8ic_conjugate_8ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_resamplerfastxnpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/asm
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/asm/orc
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_xn_resampler_16ic_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8i_max_s8i.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_x2_rotator_dotprodxnpuppet_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_x2_multiply_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32f_sincos_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_16i_rotator_dotprodxnpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_s32f_sincospuppet_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32f_index_max_32u.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_x2_rotator_dotprodxnpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_x2_rotator_dot_prod_16ic_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_x2_dot_prod_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_s32fc_x2_rotator_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8i_accumulator_s8i.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_convert_8ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_resampler_fast_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_xn_resampler_32fc_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_resamplerfastpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8i_index_max_16u.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8i_x2_add_8i.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_xn_resampler_fast_16ic_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_32f_high_dynamic_rotator_dot_prod_32fc_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_64f_accumulator_64f.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16i_xn_resampler_16i_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32f_resamplerxnpuppet_32f.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_resamplerxnpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_convert_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8u_x2_multiply_8u.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_8ic_x2_multiply_8ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_x2_dotprodxnpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_32fc_32f_high_dynamic_rotator_dotprodxnpuppet_32fc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_rotatorpuppet_16ic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_16ic_16i_rotator_dot_prod_16ic_xn.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_alloc.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_prefs.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_complex.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_common.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/saturation_arithmetic.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_avx_intrinsics.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_sse_intrinsics.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_sse3_intrinsics.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_neon_intrinsics.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_cpu.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h
-- Installing: /build/source/build/volk_gnsssdr_module/install/include/volk_gnsssdr/volk_gnsssdr_typedefs.h
[... 5 lines cut ...]
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/cmake/volk_gnsssdr/VolkGnsssdrTargets.cmake
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/cmake/volk_gnsssdr/VolkGnsssdrTargets-release.cmake
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/libvolk_gnsssdr.so.0.0.19
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/libvolk_gnsssdr.so
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/libvolk_gnsssdr.a
[... cut ...]
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/python3.11/dist-packages/volk_gnsssdr_modtool/cfg.pyo
-- Installing: /build/source/build/volk_gnsssdr_module/install/lib/python3.11/dist-packages/volk_gnsssdr_modtool/volk_gnsssdr_modtool_generate.pyo
-- Installing: /build/source/build/volk_gnsssdr_module/install/bin/volk_gnsssdr_modtool
[ 49%] Completed 'volk_gnsssdr_module'
[ 49%] Built target volk_gnsssdr_module
make: *** [Makefile:146: all] Error 2
```

Any idea why is that? Or how to fix? A full log is available here:

https://hydra.nixos.org/build/287771073/nixlog/1

carlesfernandez, 2025-02-20:
Dear @doronbehar , thanks for your report. According to the full log, it is not volk-gnsssdr what is failing. See these lines:

```
...
In file included from /build/source/src/core/system_parameters/galileo_cnav_message.cc:22:
/nix/store/xdgg8k7j39kn0vwh0nvxn7yc00ri5ag4-glog-0.7.1/include/glog/logging.h:60:4: error: #error <glog/logging.h> was not included correctly. See the documentation for how to consume the library.
   60 | #  error <glog/logging.h> was not included correctly. See the documentation for how to consume the library.
      |    ^~~~~
In file included from /nix/store/xdgg8k7j39kn0vwh0nvxn7yc00ri5ag4-glog-0.7.1/include/glog/logging.h:63:
/nix/store/xdgg8k7j39kn0vwh0nvxn7yc00ri5ag4-glog-0.7.1/include/glog/flags.h:45:4: error: #error <glog/flags.h> was not included correctly. See the documentation for how to consume the library.
   45 | #  error <glog/flags.h> was not included correctly. See the documentation for how to consume the library.
      |    ^~~~~
...

```
It seems that CMake is not finding google-glog, or not linking against it appropriately. I see that you are passing `-DGLOG_INCLUDE_DIR:STRING=/nix/store/xdgg8k7j39kn0vwh0nvxn7yc00ri5ag4-glog-0.7.1/include 
` to CMake. Maybe you can try replacing this by `DGLOG_ROOT=/nix/store/xdgg8k7j39kn0vwh0nvxn7yc00ri5ag4-glog-0.7.1/`. Please let us know if this does the trick.

I would also recommend using the `next` branch instead on `main`.

doronbehar, 2025-02-21:
We are using the latest available release of gnss-sdr - 0.0.19.1, and the suggested `-DGLOG_ROOT` cmake flag is not supported in that release. Unfortunately, I also encountered issues compiling gnsstk which seems to be a dependency of gnss-sdr in the `next` master. TBH, I'm not using gnss-sdr, I am just an SDR fan, and I encountered this failure in our CI, and hoped the solution will be easy. Hence, I won't put any more efforts into this, so feel free to close this issue if you don't experience any issues in your CI, and the code I tried to compiled is too old to be relevant.

carlesfernandez, 2025-02-21:
Ok, thanks anyway. We plan to do a release very soon that hopefully will fix this issue.