gnss-sdr/gnss-sdr #853: CMake Error Due to Missing Dependencies and Python Version Issues

tantanli, 2025-02-21:
Description: I'm encountering issues while building GNSS-SDR. When I run 
```
cmake ..
```
 the configuration fails due to missing dependencies and Python version issues. 
Below is the output from the cmake command:
```
cmake ..
-- Configuring GNSS-SDR v0.0.19.git-next-b9fe78cf0 to be built on GNU/Linux Ubuntu 24.04 (x86_64)
-- Build type set to Release.
-- Checking for module 'orc-0.4 > 0.4.22'
--   Package 'orc-0.4', required by 'virtual:world', not found
-- Checking for module 'libzmq'
--   Package 'libzmq', required by 'virtual:world', not found
-- Checking for GNU Radio Module: RUNTIME
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_RUNTIME_FOUND = TRUE
-- Checking for GNU Radio Module: PMT
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_PMT_FOUND = TRUE
-- Checking for GNU Radio Module: BLOCKS
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_BLOCKS_FOUND = TRUE
-- Checking for GNU Radio Module: FFT
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_FFT_FOUND = TRUE
-- Checking for GNU Radio Module: FILTER
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_FILTER_FOUND = TRUE
-- Checking for GNU Radio Module: ANALOG
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/libvolk.so
-- GNURADIO_ANALOG_FOUND = TRUE
-- Checking for GNU Radio Module: UHD
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-uhd.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_UHD_FOUND = TRUE
-- Checking for GNU Radio Module: ZEROMQ
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-zeromq.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- GNURADIO_ZEROMQ_FOUND = TRUE
-- Checking for GNU Radio Module: IIO
--  * INCLUDES=/usr/include
--  * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-iio.so
-- GNURADIO_IIO_FOUND = TRUE
-- Could NOT find GNSSTK (missing: GNSSTK_LIBRARY GNSSTK_INCLUDE_DIR) 
-- Performing Test has_span
-- Performing Test has_span - Success
-- Performing Test has_rotl
-- Performing Test has_rotl - Success
-- Performing Test has_std_plus_void
-- Performing Test has_std_plus_void - Success
-- Performing Test has_transform_reduce
-- Performing Test has_transform_reduce - Success
-- Performing Test has_transform_reduce_with_execution_policy
-- Performing Test has_transform_reduce_with_execution_policy - Success
-- Looking for shm_open
-- Looking for shm_open - found
-- Could NOT find VOLKGNSSSDR (missing: VOLK_GNSSSDR_LIBRARIES VOLK_GNSSSDR_INCLUDE_DIRS) 
--  volk_gnsssdr will be built along with gnss-sdr when doing 'make'.
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
-- Could NOT find Python2 (missing: Python2_EXECUTABLE Interpreter) 
    Reason given by package: 
        Interpreter: Wrong major version for the interpreter "/home/li/miniconda3/bin/python"

-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
-- Found Python: /home/li/miniconda3/bin/python3 (found version: 3.12.7)
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
-- Mako template library not found. See https://www.makotemplates.org/
--  You can try to install it by typing:
--  sudo apt-get install python3-mako
CMake Error at CMakeLists.txt:1158 (message):
  Mako templates required to build VOLK_GNSSSDR

-- Configuring incomplete, errors occurred!
```
Environment:

OS: Ubuntu 24.04 (x86_64)
Python: 3.12.7 (installed via Miniconda)
CMake: 3.28.3
I tried switching to the "next" branch using git checkout next, but I encountered the same issue both with and without switching to the "next" branch.
Thank you in advance.

tantanli, 2025-02-21:
I have installed python3-mako
```
dpkg -l | grep python3-mako
ii  python3-mako                                   1.3.2-1                                  all          fast and lightweight templating for the Python 3 platform
```

arch-dawson, 2025-02-21:
It looks like the python installation that it found is through miniconda:
`-- Found Python: /home/li/miniconda3/bin/python3 (found version: 3.12.7)`
I would try installing mako in that miniconda environment and see if that fixes the issue.

tantanli, 2025-02-21:
It works! Thank  you @arch-dawson 
I finished cmake.
But I occured another problem while I run
```
make
```
Below is the output from the make command:
```
[ 85%] Built target input_filter_adapters
[ 87%] Built target tracking_libs
[ 89%] Built target tracking_gr_blocks
[ 91%] Built target tracking_adapters
[ 92%] Built target pvt_gr_blocks
[ 92%] Built target pvt_adapters
[ 93%] Built target core_monitor
[ 94%] Built target core_receiver
[ 94%] Linking CXX executable gnss-sdr
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_global_cleanup@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_global_init@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_perform@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_init@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_slist_free_all@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_setopt@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_slist_append@CURL_OPENSSL_4'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
make[2]: *** [src/main/CMakeFiles/gnss-sdr.dir/build.make:179：src/main/gnss-sdr] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2348：src/main/CMakeFiles/gnss-sdr.dir/all] 错误 2
make: *** [Makefile:146：all] 错误 2
```
It seems similar to the issue encountered in #445 .And I tried the next branch, it didn't works.

I really need help.
Thank you.

tantanli, 2025-02-21:
Maybe this might help [#152](https://github.com/graphdeco-inria/gaussian-splatting/issues/152)

tantanli, 2025-02-22:
I finished the installer

closed 2025-02-22 by tantanli