gnss-sdr/gnss-sdr #856: SSI and LLI field/spaces missing after SNR in RINEX observation files

joebre, 2025-02-27:
SSI and LLI field or spaces are missing after the SNR observation. 
This is often tolerated for single frequency observation files but makes the RINEX observation file corrupt in case of multi frequency observable.
The RINEX 3.02 observation files are rejected by several post processing tools. The MATLAB readrinex function also fails to correctly parse the GNSS-SDR generated observation files. 

In **rinex_printer.cc** in all **Rinex_Printer::log_rinex_obs** members the following line:

 lineObs += Rinex_Printer::rightJustify(asString(iter->second.CN0_dB_hz, 3), 14);

must be followed by:

 if (lli == 0)
 {
        lineObs += std::string(1, ' ');
 }
 lineObs += Rinex_Printer::rightJustify(Rinex_Printer::asString<int32_t>(ssi), 1); 

This makes the RINEX observation file to be compliant with the standard and makes it processable with third parties tools.

Joel

Octovia7, 2025-03-02:
claim ..

Octovia7, 2025-03-02:
Hi @joebre (or maintainers),

I came across this issue and would like to work on it. Based on the description, it seems the problem is related to missing SSI and LLI fields in the RINEX 3.02 observation files generated by GNSS-SDR.

I plan to implement the suggested fix in the `rinex_printer.cc` file by adding the LLI and SSI fields after the SNR observation. Before I proceed, I wanted to confirm if this approach aligns with the intended solution.

Additionally, I have a few questions:
1. Are there any specific test cases or RINEX files I should use to verify the fix?
2. Should I handle any edge cases (e.g., when `lli` or `ssi` values are not available)?

Please let me know if there’s anything else I should consider. I’ll start working on this once I get confirmation.

Thanks!

joebre, 2025-03-04:
Hi Octavia, I personally discovered the issue by reading the the GNSS--SDR generated RINEX 3.02 observation file into MATLAB using the built-in rinexread function. MATLAB is currently unable to interpret correctly the observation file since it is not compliant with the standard. 
I'm not part of the GNSS-SDR developer team, so maybe I'm underestimating the importance of the verification process... but in this case the verification is trivial... the RINEX 3.02 standard defines that each observable must occupy exactly 14 chars (including spaces and decimal point) and must be followed by the LLI and the SSI char (spaces when not available).
The double check about the fix of the bug can be made by generating a "fixed" RINEX 3.02 musty-frequency observation file and load it int MATLAB using rinexread and verify the content of the generated timetable. Alternatively you could check the RINEX observation file integrity using tools such as GFZRNX.
Cheers, Joel

closed 2026-08-18 by joebre