gnss-sdr/gnss-sdr #860: Core Dumped error

jeaarc, 2025-03-14:
[image]
Parçalanma arızası mean Core dumped.

I'M working with Adalm pluto on Ubuntu 24.10

Octovia7, 2025-03-18:
Hi @jeaarc and everyone,

I came across this issue and wanted to share some debugging steps and potential solutions that might help resolve the segmentation fault (core dumped) with the ADALM-PLUTO SDR on Ubuntu 24.10:

Debugging Steps
Run GNSS-SDR with Debug Logs:

gnss-sdr --config_file=gnss_sdr_GPS_L1_plutosdr_realtime.conf --log_level=debug
Look for warnings or errors in the logs.

Test with a Simulated Signal:
Modify the config file to use a file source:

ini
Copy
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/path/to/gps_signal_file.dat
If it works, the issue may be hardware or driver-related.

Check PlutoSDR Firmware and Drivers:
Ensure firmware and libiio are up to date:

bash
Copy
iio_info -s
iio_attr -c -u ip:192.168.2.1 firmware_version
Analyze the Core Dump with GDB:

bash
Copy
gdb gnss-sdr core
Use bt to identify the crash location.

Check for Memory Issues with Valgrind:

bash
Copy
valgrind --tool=memcheck --leak-check=full gnss-sdr --config_file=gnss_sdr_GPS_L1_plutosdr_realtime.conf
Look for memory-related errors.

Potential Solutions
Update PlutoSDR Firmware:
Use the PlutoSDR firmware update tool if outdated.

Adjust Configuration Parameters:
Try reducing the sample rate or changing the gain mode:

GNSS-SDR.internal_fs_sps=2500000  # Reduce sample rate
SignalSource.gain_mode=manual      # Change gain mode
Check System Dependencies:
Ensure all dependencies are installed:

sudo apt update
sudo apt install libiio-dev libad9361-dev libvolk2-dev libboost-dev libgtest-dev libgflags-dev liblog4cpp5-dev libgnuradio-dev libblas-dev liblapack-dev libarmadillo-dev
Recompile GNSS-SDR:
Rebuild GNSS-SDR from source:

git clone https://github.com/gnss-sdr/gnss-sdr.git
cd gnss-sdr
mkdir build
cd build
cmake ..
make
sudo make install.

Request for Collaboration
If anyone has additional insights or suggestions, please share them. I'm happy to help test fixes or provide more information.

Octovia7, 2025-03-18:
Hi @jeaarc and everyone,

Here are some quick debugging steps and potential fixes for the segmentation fault with ADALM-PLUTO on Ubuntu 24.10:

Debugging Steps
Run GNSS-SDR with debug logs:

gnss-sdr --config_file=gnss_sdr_GPS_L1_plutosdr_realtime.conf --log_level=debug
Test with a simulated signal (file source) to rule out hardware issues.

Check PlutoSDR firmware and libiio versions:

iio_info -s
iio_attr -c -u ip:192.168.2.1 firmware_version
Analyze the core dump with gdb:

gdb gnss-sdr core
Use bt for a backtrace.

Check for memory issues with valgrind:

valgrind --tool=memcheck --leak-check=full gnss-sdr --config_file=gnss_sdr_GPS_L1_plutosdr_realtime.conf
Potential Fixes
Update PlutoSDR firmware if outdated.

Reduce the sample rate or change the gain mode in the config file.

Ensure all dependencies are installed:

sudo apt update
sudo apt install libiio-dev libad9361-dev libvolk2-dev libboost-dev libgtest-dev libgflags-dev liblog4cpp5-dev libgnuradio-dev libblas-dev liblapack-dev libarmadillo-dev
Recompile GNSS-SDR from source.

If anyone has additional insights, feel free to share! Happy to help test fixes.