gnss-sdr/gnss-sdr #868: Request for BDS B1I Baseband Signal for GNSS-SDR Configuration Verification

Autumnan7, 2025-03-25:
Hello, I would like to verify the correctness of my GNSS-SDR configuration file and observe its decoding output. Could you please advise me on where I can obtain a BDS B1I baseband signal, similar to the data file used in "My First Fix"?

Thank you very much for your help—I truly appreciate it!

vladisslav2011, 2025-03-25:
Hello.
I had the same problem some time ago with BDS B1I, B2I and B3I signal.
I've ended up recording these signals from an antenna...

Autumnan7, 2025-03-25:
> Hello. I had the same problem some time ago with BDS B1I, B2I and B3I signal. I've ended up recording these signals from an antenna...

Thank you for your help. If I can't find the data online, I’ll have to learn how to use USRP to collect signals.

hankouman, 2025-03-26:
can you give me your BDS conf  file.

vladisslav2011, 2025-03-26:
B1I realtime:
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000
GNSS-SDR.Beidou_banned_prns=56,57,58

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=4000000
SignalSource.freq=1561098000
;SignalSource.osmosdr_args=uhd,num_recv_frames=256
;SignalSource.gain=30
;SignalSource.antenna=TX/RX

;SignalSource.osmosdr_args=miri=0,bias=1,buffers=256
;SignalSource.gain=100

SignalSource.osmosdr_args=hackrf=0,bias=1,buffers=256
SignalSource.gain=0
SignalSource.rf_gain=40
SignalSource.if_gain=30

SignalSource.if_bw=4000000
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Pass_Through

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_B1.count=16
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
Acquisition_B1.item_type=gr_complex
Acquisition_B1.coherent_integration_time_ms=2
Acquisition_B1.pfa=0.000002
Acquisition_B1.doppler_max=4200
Acquisition_B1.doppler_step=100
Acquisition_B1.dump=false
Acquisition_B1.blocking=false;
Acquisition_B1.bit_transition_flag = false;

;######### TRACKING GLOBAL CONFIG ############
Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
Tracking_B1.item_type=gr_complex
Tracking_B1.extend_correlation_symbols=10
Tracking_B1.pll_bw_hz=30.0
Tracking_B1.dll_bw_hz=2.00
Tracking_B1.pll_bw_narrow_hz=15.0
Tracking_B1.dll_bw_narrow_hz=0.50
Tracking_B1.dump=false

[... 106 lines cut ...]
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=200
PVT.display_rate_ms=1000
PVT.enable_rx_clock_correction=true
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2
PVT.raim_fde=1
PVT.flag_nmea_tty_port=false
PVT.nmea_dump_devname=/dev/pts/3

PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1111

Monitor.enable_monitor=true
Monitor.decimation_factor=4
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1112
```

Autumnan7, 2025-03-27:
> can you give me your BDS conf file.

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=1300000
GNSS-SDR.pre_2009_file=false

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=
SignalSource.item_type=ishort
SignalSource.sampling_frequency=2600000
SignalSource.freq=1561098000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=2600000
Resampler.sample_freq_out=1300000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels.in_acquisition=8
Channel.signal=B1
Channels_B1.count=16

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
Acquisition_B1.item_type=gr_complex
#Acquisition_B1.coherent_integration_time_ms=1
Acquisition_B1.pfa=0.01
#Acquisition_B1.max_dwells=2
;Acquisition_B1.pfa=0.0000001;
Acquisition_B1.doppler_max=10000
Acquisition_B1.doppler_min=-10000
Acquisition_B1.doppler_step=250
#Acquisition_B1.dump=true
#Acquisition_B1.dump_filename=./acquisition/bds_acq
Acquisition_B1.blocking=true;
Acquisition_B1.bit_transition_flag = false;

;######### TRACKING GLOBAL CONFIG ############
Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
Tracking_B1.item_type=gr_complex
Tracking_B1.pll_bw_hz=40.0;
Tracking_B1.dll_bw_hz=4;
#Tracking_B1.extend_correlation_symbols=10
#Tracking_B1.cn0_min=27
;Tracking_B1.cn0_min=1
#Tracking_B1.dump=true
#Tracking_B1.dump_filename=./tracking/epl_tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_B1.implementation=BEIDOU_B1I_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

Autumnan7, 2025-03-27:
> My working config files if anyone want to know. B1I realtime:
> 
> ```
> [GNSS-SDR]
> 
> ;######### GLOBAL OPTIONS ##################
> GNSS-SDR.internal_fs_sps=4000000
> GNSS-SDR.Beidou_banned_prns=56,57,58
> 
> ;######### SIGNAL_SOURCE CONFIG ############
> SignalSource.implementation=Osmosdr_Signal_Source
> SignalSource.item_type=gr_complex
> SignalSource.sampling_frequency=4000000
> SignalSource.freq=1561098000
> ;SignalSource.osmosdr_args=uhd,num_recv_frames=256
> ;SignalSource.gain=30
> ;SignalSource.antenna=TX/RX
> 
> ;SignalSource.osmosdr_args=miri=0,bias=1,buffers=256
> ;SignalSource.gain=100
> 
> SignalSource.osmosdr_args=hackrf=0,bias=1,buffers=256
> SignalSource.gain=0
> SignalSource.rf_gain=40
> SignalSource.if_gain=30
> 
> SignalSource.if_bw=4000000
> SignalSource.AGC_enabled=false
> SignalSource.samples=0
> SignalSource.repeat=false
> SignalSource.enable_throttle_control=false
> SignalSource.dump=false
> SignalSource.dump_filename=./signal_source.dat
> 
> ;######### SIGNAL_CONDITIONER CONFIG ############
> SignalConditioner.implementation=Signal_Conditioner
> 
> ;######### DATA_TYPE_ADAPTER CONFIG ############
> DataTypeAdapter.implementation=Pass_Through
> 
> ;######### INPUT_FILTER CONFIG ############
> InputFilter.implementation=Pass_Through
> 
> ;######### RESAMPLER CONFIG ############
> Resampler.implementation=Pass_Through
> 
> ;######### CHANNELS GLOBAL CONFIG ############
> Channels_B1.count=16
> Channels.in_acquisition=1
> 
> ;######### ACQUISITION GLOBAL CONFIG ############
> Acquisition_B1.implementation=BEIDOU_B1I_PCPS_Acquisition
> Acquisition_B1.item_type=gr_complex
> Acquisition_B1.coherent_integration_time_ms=2
> Acquisition_B1.pfa=0.000002
> Acquisition_B1.doppler_max=4200
> Acquisition_B1.doppler_step=100
> Acquisition_B1.dump=false
> Acquisition_B1.blocking=false;
> Acquisition_B1.bit_transition_flag = false;
> 
> 
> ;######### TRACKING GLOBAL CONFIG ############
> Tracking_B1.implementation=BEIDOU_B1I_DLL_PLL_Tracking
> Tracking_B1.item_type=gr_complex
> Tracking_B1.extend_correlation_symbols=10
> Tracking_B1.pll_bw_hz=30.0
> Tracking_B1.dll_bw_hz=2.00
> Tracking_B1.pll_bw_narrow_hz=15.0
> Tracking_B1.dll_bw_narrow_hz=0.50
[... 328 lines cut ...]
> PVT.display_rate_ms=1000
> PVT.enable_rx_clock_correction=true
> PVT.flag_rtcm_server=true
> PVT.flag_rtcm_tty_port=false
> PVT.rtcm_dump_devname=/dev/pts/1
> PVT.rtcm_tcp_port=2101
> PVT.rtcm_MT1019_rate_ms=5000
> PVT.rtcm_MT1077_rate_ms=1000
> PVT.rinex_version=2
> PVT.raim_fde=1
> PVT.flag_nmea_tty_port=false
> PVT.nmea_dump_devname=/dev/pts/3
> 
> 
> PVT.enable_monitor=true
> PVT.monitor_client_addresses=127.0.0.1
> PVT.monitor_udp_port=1111
> 
> Monitor.enable_monitor=true
> Monitor.decimation_factor=4
> Monitor.client_addresses=127.0.0.1
> Monitor.udp_port=1112
> ```

Thank you for your .conf. Do you have any `.dat` files related to B1I?