gnss-sdr/gnss-sdr #871: GNSSTK not found

wkazubski, 2025-04-08:
It seems that GNSSTK is now necessary to build gnss-sdr. On openSUSE the prebuilt GNSSTK version 14.3.0  is available as a package from Application:Geo repository, but gnss-sdr cannot find it. 

The gnsstk -devel package installs Cmake files to:
/usr/lib64/cmake/GNSSTK
or /usr/lib/cmake/GNSSTK (for 32-bit architectures)
and library headers to:
/usr/include/gnsstk14/gnsstk
Libraries are at standard location /usr/lib64 or /usr/lib.

carlesfernandez, 2025-04-09:
Dear @wkazubski ,

GNSSTK is not required for building gnss-sdr, it's only for some optional tests. For instance:

```
cmake -DENABLE_PACKAGING=ON -DENABLE_UNIT_TESTING_MINIMAL=ON -DENABLE_OSMOSDR=ON -DENABLE_FMCOMMS2=ON -DENABLE_PLUTOSDR=ON -DENABLE_RAW_UDP=ON -DENABLE_LIMESDR=ON -DENABLE_ZMQ=ON ..
```
builds most of the optional signal sources but it does not require GNSSTK.

wkazubski, 2025-04-16:
Thanks for help.
After adding -DENABLE_UNIT_TESTING_MINIMAL=ON, gnss-sdr builds for several frontends, like Osmosdr or my frontend with MAX2769, but not for others like PlutoSDR. In each case cmake searches for gnsstk and fails with message that GNSSTK headers are not found. Adding -DGNSSTK_INCLUDE_DIR=/usr/include/gnsstk14 does not help.

carlesfernandez, 2025-04-17:
It seems that you are activating a building option that requires GNSSTK, please check https://gnss-sdr.org/docs/tutorials/configuration-options-building-time/ for documentation.

Anyway, I pushed a workaround to the `next` branch (see 6097c0f5119f760a1ad7c8bb1bd139a5683b250a) that should fix the finding of the gnsstk-devel package, which uses that weird `/usr/include/gnsstk14/gnsstk` path for the header files. Hope this helps.

wkazubski, 2025-04-17:
Now I even got two extra binary files gps_l1_ca_dll_pll_tracking_test_fpga and obsdiff. I thik that tey are for testing.

I found that someone activated the option -DVERSIONED_HEADER_INSTALL while building gnsstk. This option adds 
extra directory level to include files.

closed 2025-04-18 by wkazubski