gnss-sdr/gnss-sdr #873: `cmake` version 4.0.0: Error `Compatibility with CMake < 3.5 has been removed from CMake.`.

dreirund, 2025-04-13:
Ahoj,

with `cmake` version 4.0.0 I get the following error:  
```
[...]
Cloning into 'gnss-metadata-standard'...
HEAD is now at 220d116 Merge branch 'master' of https://github.com/IonMetadataWorkingGroup/GNSS-Metadata-Standard
CMake Error at /var/cache/makepkg/build/gnss-sdr-git/src/build/thirdparty/gnss-metadata-standard/CMakeLists.txt:32 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
```

Adding `-DCMAKE_POLICY_VERSION_MINIMUM=3.5` does help.

jwmelto, 2025-04-13:
CMake 4.0 has some changes that break backwards compatibility. This project values supported environments over being over being on the bleeding edge. 

Support for CMake 4.0 will be slow in coming, I think

carlesfernandez, 2025-04-14:
Hi @dreirund 

thanks for your report. Since that repository seems to be not maintained, I pushed a workaround to our `next` branch at  fa491bbf338cf7a2f35c106c1301aab65dbba781. You will need to start from a fresh building folder and with  the `patch` command installed.

dreirund, 2025-04-14:
> *Since that repository seems to be not maintained*

Why is this not maintained?

[Last commit](https://github.com/gnss-sdr/gnss-sdr/commits/main/) seemed to have been on 2025-04-02, so pretty recently.

Regards!

carlesfernandez, 2025-04-15:
I was referring to https://github.com/IonMetadataWorkingGroup/GNSS-Metadata-Standard , which is downloaded and built if `-DENABLE_ION=ON` is passed to CMake.