gnss-sdr/gnss-sdr #881: PVT time latency

Derso1, 2025-04-25:
Hello everyone

I am trying to measure the time between point where I received RF information until the point that I have the consequent PVT release. 
I know the point inside de code to put my final probe (PVT release), but I don’t know the initial point (RF input or RF sample) inside the GNSS-SDR code.

Somebody, Can, please, help-me?

jwmelto, 2025-04-25:
There's an old proverb that says "a man with one watch always knows what time it is. A man with two watches is never sure." There are two things to consider in your problem. The first is what watch you use (and how accurate it is). The second is where (when)  you measure the time of event.

A related problem is knowing what you intend to do with these time measurements. Presumably, you are looking to measure the latency between time of intercept and time of solution. For GNSS, there are fundamental limitations that enforce a certain amount of latency (how long it takes to transmit a complete frame, or 4 nav messages based on the GNSS-SDR implementation), as well as the dependency on collecting multiple satellites.

Another factor that could affect your calculation of latency is how many threads you use for acquisition and tracking; fewer threads will force a serialization of the process of dealing with each tracked satellite. Processing power can also be an impact. But perhaps that's precisely what you are trying to measure.

A good reference for understanding when (where) to measure time is the VITA 49 specification. When a receiver reports time, it must explain at which point the time is measured. Possible places for measuring time include:
- at the antenna (point of reception)
- at the point of digitization
- at the point of transmission

This list isn't exhaustive, but is illustrative. 

Now, in GNU Radio, "time" isn't really supported (directly). The _passage_ of time is inferred by the number of samples in the stream, but absolute time is not asserted. That is, _if_ you know the sample rate and you know the number of samples, then you can compute the amount of elapsed time.

How accurately you need to measure the initial point depends on what you want to accomplish with this elapsed time calculation. From a GNSS-SDR perspective, the "entry point" would be the signal source, but given the multiple supported signal sources, creating a general purpose timestamp is problematic. Some radios provide a measured time of intercept, perhaps as a tag in the GNU Radio data stream. It should be possible to add time-of-intercept tags to a data stream in a particular signal source, if you like.

The real challenge, though, is how to tell which sample correlates to your PVT solution. By the time you get to PVT, the input stream has been transformed into "observables", and the direct correlation to specific samples is probably lost.

GNU Radio does try to propagate tags coherently from input to output, even if there is a translation (think decimation) of the number of inputs to the number of outputs. However, GNSS-SDR has some brittle tag handling code, so it doesn't do well with unexpected tags in the data stream. At this time, I don't recommend trying to add tags into the data stream. In fact, in a few places tag propagation is explicitly disabled, because random tags cause problems.

All of that is to say, I don't think there's a general solution to your problem. You might try measuring the time at the point of acquisition, but that's relative to a specific satellite; you need at least 4 satellites to get a solution.

All this is to say, you asked a very hard question, without enough context to provide a useful answer (assuming such an answer exists).

Derso1, 2025-04-25:
Hello @jwmelto 
Thank you for your extended answer and I can complete her with another proverb “If you don't know where you're going, any road will get you there” (that is the picture of my situation).

Anyway, I already understood that the complete way from RF until PVT solution is a utopic question, but you put:

“… might try measuring the time at the point of acquisition, but that's relative to a specific satellite; you need at least 4 satellites to get a solution.”

Ok, It is not the complete way, but considering the point of acquisition, Can you guide-me where it is the best moment to try insert my start point of measure?

I am using a Hack RF. Do you know if there is some measured time of intercept in the GNU-Radio for this device?

Thanks again and a lot.