gnss-sdr/gnss-sdr #883: Next branch error with GCC14

ph4ntome, 2025-04-29:
Source cannot compile cause of
` fft_kernelstring.cc:894:13: error: format not a string literal and no format arguments snprintf((kInfo)->kernel_name, sizeof((kInfo)->kernel_name), kernelName.c_str());

fft_kernelstring.cc:1146:21: error: format not a string literal and no format arguments snprintf((kInfo)->kernel_name, sizeof((kInfo)->kernel_name), kernelName.c_str()); `

As I understand GCC 14 has gotten stricter about the use of snprintf() when the format string is not a literal.

I have not tried it yet but replacing

`snprintf((kInfo)->kernel_name, sizeof((kInfo)->kernel_name), kernelName.c_str());`

with 

`strncpy((kInfo)->kernel_name, kernelName.c_str(), sizeof((kInfo)->kernel_name));
(kInfo)->kernel_name[sizeof((kInfo)->kernel_name) - 1] = '\0';`

would solve it I think.

jwmelto, 2025-04-30:
wow. I had no idea there was any OpenCL support at all. Looking at that function, I'm still not.

What was there was horribly broken (`sizeof(*kInfo)->kernel_name)` evaluates to `sizeof(char*)` which is certainly not what was meant).

I believe this needs to be enabled with `-DENABLE_OPENCL=ON`; the top-level CMakeLists.txt identifies it as experimental. The simplest fix is to not enable it.

ph4ntome, 2025-04-30:
Yeah default disable seems to be the best solution here. Have a wonderful day and thanks for your time.