gnss-sdr/gnss-sdr #901: Driving GNSS-SDR from multiple sources - only one source is working

rogerandersen, 2025-07-11:
I'm trying to run GNSS-SDR with multiple inputs, one for L1, another for L5.
I've got the data in two separate files.
I am supposedly following the guidance for this presented online, here [https://gnss-sdr.org/docs/sp-blocks/signal-source/#multiple-sources], for configuring for two input sources.
However, when I run, the SDR seems to ignore the second L5 input and just operate as if I was only configured for L1.
Can anyone suggest what my config file is lacking here?
I'll attach it for reference.

[cfgL1L5-sky+.txt](https://github.com/user-attachments/files/21189482/cfgL1L5-sky%2B.txt)

Teja2007-max, 2025-07-15:
This happens because of  GNSS SDR supports the multiple sources but require for the correct mapping btw:
1)Signal Sources 
2)Channels
3)Signal processing blocks
here  are the some reasons for the blocking and lagging of the L5
1)//SignalSource.1 block is missing
2)//Channel is placed to the wrong source id
3)//Sample frequency mismatch
4) //L5 acquisition/tracking blocks missing and etc.
Take the two different inputs sources
and try for the different frequencies of sps
 Define L5 as a separate Signal Source with a unique source_id
now  assign  the values for this
I will keep the discription and  I'll attach reference below

[Final Working Solution Use L1 + L5.txt](https://github.com/user-attachments/files/21225067/Final.Working.Solution.Use.L1.%2B.L5.txt)

rogerandersen, 2025-07-17:
> This happens because of GNSS SDR supports the multiple sources but require for the correct mapping btw: 1)Signal Sources 2)Channels 3)Signal processing blocks here are the some reasons for the [...]

I appreciate the response.
However, it confuses me in several ways.
- The L5 acquisition and tracking blocks were not missing - see lines 110 and 130.
- The SignalSource for channel 1 was not missing - see line 16.
- The posted solution uses "SignalSource.0" and "SignalSource.1" rather than "SignalSource0" and "SignalSource1". That ".0" and ".1" syntax isn't reflected in the documentation, and the config file examples in the repo don't contain any examples of it.
- The posted solution also contains no signal conditioning elements.

So it's not clear how to make use of this suggested solution.

Teja2007-max, 2025-07-19:
Could you pls send me the source code for this .
Because I am out of station and I haven't bought my system
Once me the code so that i can reply with in an hour
if possible I will reply with in a hour for the query
you can communicate with the same mail id

Teja2007-max, 2025-07-20:
I have tried my level best to this problem and I gave the all efforts to solve this problem 
I will the reference document to verify .
If there are any issues your facing you can mail me 

[Issue 1 L5 acquisitiontracking bloc.txt](https://github.com/user-attachments/files/21333309/Issue.1.L5.acquisitiontracking.bloc.txt)