gnss-sdr/gnss-sdr #917: Huge pseudo-distances not consistent between sdr and gnss receiver

carolim-git, 2025-09-03:
Hi everyone,
I am quite new to gnss-sdr, and would very much appreciate your help on a matter I've been struggling with for a while now.
I am using an SDR X310 to receive GPS L1 C/A signal from a GNSS simulator, always using the same satellites scenario for this experiment. 
For comparison, I am using a PolaRx5 receiver, with the same scenario. I use the same cables and references (PPS and 10 MHz outputs from the simulator, which is locally referenced).

I compute 
- the delay D1 between the simulator and PolaRx5
- the delay D2 between the simulator and x310

D1 and D2 should be very similar, but I obtain :
- D1 : an average 56.113 ns delay (16.822 m pseudo distance), with less than 500 ps fluctuation, for all satellites 
- D2 : fluctuating delays/pseudo distances with gnss-sdr, with an average delay offset by few ms compared to D1 (few 1000 km !), and with few ns (sometimes 10 ns) offsets between the satellites ...

Do you have any idea why this is behaving like that ? Thank you very much in advance for your help.

--------------------
Here is typically the configuration file I am using :

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source 
SignalSource.device_address=192.168.10.2  
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.clock_source=external 
SignalSource.freq=1575420000
SignalSource.subdevice=A:0
SignalSource.repeat=false ; default, makes only sense for file
SignalSource.dump=false
SignalSource.enable_throttle_control=false 

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Pass_Through ; copies samples from input to output

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=12
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10 ; the higher the number, the better the clock stability -> no problem for us, so try out, before 20 (default is 1)

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single ; there is also PPP available
PVT.output_rate_ms=30000 ; change: before it was 500, then 1000
PVT.display_rate_ms=1000
PVT.elevation_mask=10; default=15
PVT.iono_model=off;
PVT.geojson_output_enabled=true
PVT.output_enabled=false
PVT.rinex_output_enabled=true

carolim-git, 2025-09-03:
_Also I am sorry in advance if that issue was raised before, I have dug into github issues but didn't find the answers I am looking for_

chengjiedavy, 2025-09-21:
You can try setting GNSS-SDR's pre-coherent bandwidth and correlation spacing to be the same as the PolaRx5, and see if the problem improves.